intel/zephyr.js #269: [test-buffer.js] Unable to allocate buffer when passing an unsigned long value(2^32 - 1)

xiaolongx, 2016-10-10:
passing a readonly unsigned long value to buffer generator, fail to create buffer object. 
The test case failed when the buffer length large than 256.

Commid id | Result
--- | ---
c21b9fe79926732da25c78f6c6883c96f974b72d| Pass
147bb63c305ea58d94ed1bf6689e7352eff1a03f| Fail 

Log info:
PASS - The length of Buffer(256) expected:256 got:256
zjs_buffer_create: unable to allocate buffer
FAIL - The length of Buffer(257) expected:257 got:undefined
zjs_buffer_create: unable to allocate buffer
FAIL - The length of Buffer(4294967295) expected:4294967295 got:undefined
zjs_buffer_create: unable to allocate buffer
FAIL - The length of Buffer(4294967296) expected:4294967295 got:undefined

grgustaf, 2016-11-28:
Obviously it was never going to work to have a 4GB buffer on a machine with 80KB of RAM. :)

Someone should dig into how exactly this got limited to 256 so we understand and document it, but really that is probably fine.

xiaolongx, 2016-12-06:
This issue cannot reproduce with commit a162db1 on both Arduino_101

closed 2017-01-09 by xiaolongx