intel/zephyr.js #277: Fail to find Arduino 101 via BLE in WebBluetoothDemo.js with commit 147bb63

xiaolongx, 2016-10-11:
After launching [WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js) and [WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js), nRF-Connect and Physical Web fail to find Arduino 101 using Nexus 5.
I found this issue occurs after commit(a06678264c86fe70f2cd2d34b979f80e159c8f5c) merged.

Commit | Date | Result
--- | --- | ---
147bb63c305ea58d94ed1bf6689e7352eff1a03f | Oct 7 | Fail
5466a03acc9b8ec93547de35aa0a1347cde42ce3 | Oct 5 | Pass

kenchris, 2016-10-11:
The web apps finds the device here but it cannot connect

jimmy-huang, 2016-10-11:
I don't see how the last commit can be causing issues, I think we have a bug in our BLE code in an older commit somewhere, I'll do some debugging.

jimmy-huang, 2016-10-11:
i forgot i lend geoff the nexus phone, i'll try to see if i can borrown another phone

jimmy-huang, 2016-10-11:
btw @xiaolongx when you see it fails, do you see the following all 4 lines printed on the console?  When I flashed it this morning with master, i wasn't seeing the last 2 lines, but then later I rebuild and reflashed both arc and x86, and then i can see all 4 lines.

ipm_console0: 'Sensor core running ZJS ARC support image'
WebBluetooth Demo with BLE...
Subscribed to temperature change
Advertising as physical web device

xiaolongx, 2016-10-12:
Yes, the latest code only print 2 lines, but I try to rebuild and reflash both arc and x86 many times, there are still only 2 lines:

ipm_console0: 'Sensor core running ZJS ARC support image'
WebBluetooth Demo with BLE...

And I meet a strange case, switch to commit 5466a03acc9b8ec93547de35aa0a1347cde42ce3(which everything work well) firstly, build and flash WebBluetoothDemo.js, there are 4 lines printed. Then switch back to commit 147bb63c305ea58d94ed1bf6689e7352eff1a03f, rebuild and reflash both arc and x86, there will be 4 lines printed. Is there anything cached in memory? When I unplug and plug the power(AM-BM line) to restart it, it will only print 2 lines again.

jimmy-huang, 2016-10-13:
The problem is that when we added the 'this' object patch a06678264c86fe70f2cd2d34b979f80e159c8f5c, we didn't acquire the 'this' object when registering the C callbacks, so randomly the 'this' object gets garbage-collected by JerryScript, and when zjs_callback is calling the JS function will ends up with an error, but at the same time we are not actually checking errors on returns, @jprestwo just submitted a patch to fix this, @xiaolongx can you check if #289 fix this issue? I tried it on mine and I no longer run into this issue.

xiaolongx, 2016-10-14:
#289 fix this issue,  I can see all 4 lines. The [WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js) and [WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js) work well via Physical Web.

ipm_console0: 'Sensor core running ZJS ARC support image'
WebBluetooth Demo with BLE...
Subscribed to temperature change
Advertising as physical web device

closed 2016-10-14 by xiaolongx