intel/zephyr.js #282: frdm_k64f build is broken

pfalcon, 2016-10-11:
With the current master (147bb63c30), trying to follow README instructions:

~~~
$ make BOARD=frdm_k64f JS=samples/HelloWorld.js
...
  CC      src/main.o
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/main.c: In function 'main':
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/main.c:85:15: error: implicit declaration of function 'strnlen' [-Werror=implicit-function-declaration]
         len = strnlen(script_gen, MAX_SCRIPT_SIZE);

~~~

jprestwo, 2016-10-11:
This should be a simple include fix (string.h I think) but I am seeing different errors when I build:
```
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/libexec/arm-poky-eabi/gcc/arm-poky-eabi/5.2.0/real-ld: error: .tmp_zephyr.prebuilt uses VFP register arguments, /opt/zephyr-sdk/sysroots/armv5-poky-eabi/usr/lib/./libm.a(lib_a-s_nextafter.o) does not
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/libexec/arm-poky-eabi/gcc/arm-poky-eabi/5.2.0/real-ld: failed to merge target specific data of file /opt/zephyr-sdk/sysroots/armv5-poky-eabi/usr/lib/./libm.a(lib_a-s_nextafter.o)
```

I have not built for the K64 in a long time so I am not sure if there is anything I need to do before hand to build correctly. Any suggestions?

pfalcon, 2016-10-11:
I'm going to submit a fix shortly, similar to fix we with @sergioamr applied to jerryscript itself for a similar issue.

pfalcon, 2016-10-11:
> I have not built for the K64 in a long time so I am not sure if there is anything I need to do before hand to build correctly. Any suggestions?

So, with https://github.com/01org/zephyr.js/pull/283 , everything builds and runs for me. But the error above is familiar - some time ago, Zephyr upstream was changed to use Cortex-M4F optimizations for FRDM-K46F. The corresponding changes were propagated to JrS, and per above, everything works for me with a fresh zephyr.js setup. So, I may only imagine you may have stale Zephyr/JrS checkout somewhere.

poussa, 2016-10-12:
Fixed with #283

closed 2016-10-12 by poussa

xiaolongx, 2016-10-13:
Verified k64f can build in master branch, but #283 is still missing in devel branch.
@pfalcon, could you submit this PR to devel branch?

pfalcon, 2016-10-13:
I don't think it makes much sense, it can be cherry-picked by a maintainer.

poussa, 2016-10-13:
Yeah, we'll pick it up. No need to submit.