intel/zephyr.js #284: Current pin handling design doesn't promote code portability and reuse

pfalcon, 2016-10-11:
From docs:

arduino101_pins:
LED0 controls an onboard green LED.
LED1 controls another onboard green LED. It is active low.
LED2 controls an onboard red fault LED. It is active low.

k64f_pins:
LEDR controls the red portion, LEDG the green portion, and LEDB the blue portion. They are all active high.

It's even not clear why they're named arduino101_pins, k64f_pins. It's not like someone running on arduino101 will want to use k64f's pins. An obvious solution is to name the module board_pins, and make its content be specific to a board for which Z.js is built, why following some consistent conventions for naming LEDs.

poussa, 2016-10-14:
Yeah, this is a known issue which @grgustaf has been planning to fix. Thanks for the report. We'll start addressing it.

grgustaf, 2016-12-05:
@pfalcon are you satisfied with the state of things now that your patch is merged, or do you see further steps that are needed at this time?

I'm not sure I agree about naming the different board modules the same thing, because the code you write (pin names) is different. If the interface was the same, like a "Generic Arduino" module that I've thought of making, then yes it would substitute the right back end at build time.

pfalcon, 2016-12-05:
So, there're 2 sides of this issue, largely orthogonal:

1. Being able to use Z.js on any Zephyr board without making extra changes on Z.js side. This is tracked via #368, and there's still a small change to do, that's why I don't close it.
2. Another issue is, having made it possible to run Z.js on any board, make it possible to run the same JavaScript script on any (as many as possible) boards without many (as few as possible) changes.

This ticket is dedicated to 2nd issue.

> I'm not sure I agree about naming the different board modules the same thing, because the code you write (pin names) is different.

So, one of the ideas would be to make the pin names *the same*. For example, if a board has a blue LED, that there may be a convention that it should be named "LED_BLUE", and nothing else.

All in all this boils down to a question: Is one of the aims of Zephyr.js is to allow to write portable applications in JavaScript, which would run across different boards, or there's no such aim? I'd personally very much appreciate the answer to this question - it allows to understand Zephyr.js' paradigm better (and then contribute more effectively). Thanks.

pfalcon, 2016-12-05:
Ok, the problem can be approached from the other end: JerryScript itself builds fine, so the problem must be in how Zephyr.js calls it to build. And well, it uses @nashif's way to build 3rd-party libs, as examplified by https://github.com/nashif/zephyr_modules. With this approach, Zephyr calls 3rd-party build system from its (Zephyr's) context.

By there was a counter-argument against that approach (discussed in https://jira.zephyrproject.org/browse/ZEP-366 and patches linked from it): 3rd-party build systems may be arbitrary complex, and it may be cumbersome to try to integrate them "into Zephyr". On the other hand, Zephyr's build system while complex, from our point of view it's at least "finitely complex", so using "inversion of control" approach would be better, where 3rd-party build system calls Zephyr to: 1) get build config (CC, CFLAGS); 2) build Zephyr source; 3) perform final link, with 3rd-party app injecting its own libs into it. (Steps 2&3 can be a single stage.)

This approach is now fully implemented in Zephyr upstream (https://gerrit.zephyrproject.org/r/#/c/4915/) and even documented there (https://gerrit.zephyrproject.org/r/6951). That's also what projects like JerryScript and MicroPython use.

Back to the topic of this ticket, Zephyr, via Makefile.app, calls JrS build system. But that's not how it's expected to be called, it expects to call Zephyr itself. When it does so (it's recursive call back into Zephyr build system thus), something goes wrong.

So, the solution is to call JrS makefile directly, instead of "passing it as a callback" to Zephyr. This seems to work well, after additional testing I'll submit a patch.

And indeed, it calls it pretty interesting way: it uses undocumented (within bounds of Zephyr tree) magic called "Makefile.app" which was @nashif's try on supporting linking 3rd-party libs with Zephyr.

pfalcon, 2016-12-05:
(There was a misplaces comment posted, then removed, sorry for the noise.)

grgustaf, 2016-12-07:
I'm sorry, I was going through all the issues yesterday so I was skimming a bit and I see I misread this one. :) So yes, you're really talking about the case of making common code across Arduino platforms and I agree.

Tangentially, I had a more out of the box idea recently I'll bring up again here. We could *not* specify pins at all but instead make resource requests and then at compile time have ZJS spit out a report about how to wire up your particular board (or perhaps tell you there are not enough available PWM pins, etc). That seemed interesting but a rather different approach.

So back to the topic, yes I do think it's a goal to have common code across boards whenever possible.

pfalcon, 2016-12-07:
> We could not specify pins at all but instead make resource requests and then at compile time have ZJS spit out a report about how to wire up your particular board (or perhaps tell you there are not enough available PWM pins, etc). That seemed interesting but a rather different approach.

Indeed. I mentioned on IRC my idea how to approach it without committee-like decisions on naming etc. (which likely won't scale as more boards are being added (inside or outside the main project)): split each application to at least 2 modules: the main app and "hardware config". Consider for example current `samples/arduino/basics/Blink-k64f.js`. From it,

~~~
var pins = require("k64f_pins");
var pin = gpio.open({
    pin: pins.LEDG,
    direction: 'out'
});
~~~

, would be split into hwconfig.js. Then Blink-k64f.js would do:

~~~
hwconfig = required("hwconfig.js");
...
setInterval(function () {
    toggle = !toggle;
    hwconfig.pin.write(toggle);
}, 1000);
~~~

This way, the main app (which may be arbitrary complex) stays intact, and a user will need to edit only HW config for their particular board. For example, with https://github.com/01org/zephyr.js/pull/438, for BOARD=96b_carbon it will be:

~~~
var gpio = require("gpio");
var pin = gpio.open({
    pin: "GPIOA.15",
    direction: 'out'
});
~~~

I guess, put like that, my idea is somewhere mid-way between what's currently available and yours. And well, I already try to exercise it in another project, so if Z.js follows another approach, it would only make it more interesting.

grgustaf, 2017-03-24:
Sorry this is a hard one I keep meaning to get to. I will try to make it a focus after 0.3 release.

grgustaf, 2017-07-26:
I attempted to make some progress in this direction with commit bd07595, but haven't gotten back to it now in quite a while. We now have a require("board").name property so one way to do board-specific initialization is to test board.name. @brianjjones just recently made an example of this in SPI.js with commit bac5977.

Now again we're trying to stabilize for a 0.4 release so further progress will probably wait again; sorry it's so slow.

grgustaf, 2017-12-18:
I now pretty much have this where I want it, to be able to close this issue.

We made some progress on this long ago, but the pieces that have fallen into place lately are:
- #1562 added support for PWM through board module instead of board-specific pin modules
- #1716 added the same for AIO and removed those pin modules entirely
- #1727 which I just submitted moves friendly "board support" into standalone .h files
- #1728 adds documentation of "basic board support" for boards w/o such a file, and also explains how such support could be added.

closed 2017-12-18 by grgustaf