intel/zephyr.js #285: "docs" subdir isn't mentioned in README

pfalcon, 2016-10-12:
"docs" subdir isn't mentioned anywhere in README, even though there's a special "Subdirectories" section. I guess it makes sense to mention it to minimize chance that users miss it, as well as gives hints how it's supposed to be used (is viewing online on Github the recommended way to read them)?

To not create a separate issue, there's a misplaced/weird line at the end of "Subdirectories" secion https://github.com/01org/zephyr.js#subdirectories : "The command will display the current expected partition size to remind you of the expected target."

poussa, 2016-10-14:
Good catch. Need to be fixed. First PR wins!

pfalcon, 2016-10-14:
Ok, my try is at https://github.com/01org/zephyr.js/pull/292.

poussa, 2016-10-14:
Fixed

closed 2016-10-14 by poussa