intel/zephyr.js #295: [Buffer] Precise data bus error in K64F

xiaolongx, 2016-10-17:
After flashing [Blink-k64f.js](https://github.com/01org/zephyr.js/blob/devel/samples/arduino/basics/Blink-k64f.js) and rebooting frdm_k64f, precise date bus error occus

```
Error log:
Buffer stress test...
Allocating buffer 1
Allocating buffer 2
Allocating buffer 3
***** BUS FAULT *****
  Executing thread ID (thread): 0x20007088
  Faulting instruction address:  0x0001913c
  Precise data bus error
  Address: 0x00f41ccc
Fatal fault in task ! Aborting task.
```

poussa, 2016-10-17:
Which branch and commit ID you have?

xiaolongx, 2016-10-17:
In devel branch.
Commit ID: aa0ba22297bf0c6b9a4be867854a7ee084737270

grgustaf, 2016-12-05:
Based on my comment in #297 I guess that is now commit 81a0351 on master.

brianjjones, 2016-12-07:
This appears to be fixed on master, I ran Blink-k64f.js for several minutes without any issue.  @xiaolongx please retest with the master branch

xiaolongx, 2016-12-08:
Sorry for the confusion, update the description, please take a look.

brianjjones, 2016-12-13:
@xiaolongx  BufferMemory.js  works for me on the current master branch.  I see it count up to 1000 with no issue.

xiaolongx, 2016-12-13:
This issue only reproduce on FRDM_K64F, pass on Arduino 101.
But now, it blocked by #524 after commit dc703a1.

brianjjones, 2016-12-14:
Sorry I wasn't clear, but I tested sing the FRDM_K64 and had no issues.

xiaolongx, 2016-12-15:
This issue cannot reproduce after #535 merged on FRDM-K64F, close.

closed 2016-12-15 by xiaolongx