intel/zephyr.js #296: Fail to build I2C.js and GroveLCD.js with BOARD=frdm_k64f

xiaolongx, 2016-10-17:
Unable to build [I2C.js](https://github.com/01org/zephyr.js/blob/devel/samples/I2C.js) and [GroveLCD.js](https://github.com/01org/zephyr.js/blob/devel/samples/GroveLCD.js) on commit aa0ba22297bf0c6b9a4be867854a7ee084737270 in devel branch.
please refer to the error logs:

```

I2C.js:
src/built-in.o:(.data.zjs_modules_array+0x4): undefined reference to `zjs_i2c_init'
collect2: error: ld returned 1 exit status
/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr/Makefile:810: recipe for target '.tmp_zephyr.prebuilt' failed
make[3]: *** [.tmp_zephyr.prebuilt] Error 1
make[3]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js/outdir'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr'
/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr/Makefile.inc:75: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js'
Makefile:23: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2

GroveLCD.js
src/built-in.o:(.data.zjs_modules_array+0x4): undefined reference to `zjs_grove_lcd_init'
collect2: error: ld returned 1 exit status
/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr/Makefile:810: recipe for target '.tmp_zephyr.prebuilt' failed
make[3]: *** [.tmp_zephyr.prebuilt] Error 1
make[3]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js/outdir'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr'
/home/xiaolong/00_xiaolong/test/1/zephyr.js/deps/zephyr/Makefile.inc:75: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/xiaolong/00_xiaolong/test/1/zephyr.js'
Makefile:23: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2
```

jimmy-huang, 2016-10-17:
I am not sure if I2C is actually supported on the frdm_k64f, at least not the way how arduino does it with the ARC and X86 cores, @brianjjones have you actually tried this?  This might not be a valid test case and we should add a build warning check or something

hanyia, 2016-11-08:
@brianjjones and @jimmy-huang  could you help to confirm frdm_k64f support I2C.js and GroveLCD.js?

grgustaf, 2016-11-08:
It's definitely not going to work right. We have not begun to look at supporting this in a reasonable way across multiple boards. Everything that is handled by the ARC on A101 is written incorrectly to be able to work on any other board at this point. Good reminder...

haoxli, 2016-11-08:
@xiaolongx please close this issue

closed 2016-11-09 by xiaolongx