intel/zephyr.js #297: Fail to find frdm_k64f via BLE with commit aa0ba22 in devel branch

xiaolongx, 2016-10-18:
After flashing [BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js) with commit aa0ba22297bf0c6b9a4be867854a7ee084737270, nRF-Connect and Physical Web fail to find frdm_k64f using Nexus 5

poussa, 2016-10-18:
Is this a regression? Did it work prior to the commit?

haoxli, 2016-10-18:
Not a regression, we test it first time on K64F after K64F can build successfully.

grgustaf, 2016-10-18:
BTW I had to rebase devel to merge it onto master, so the commit aa0ba22 is now 81a035 on master.

grgustaf, 2016-12-05:
Oh, there is no BLE hardware on the K64F nor has anyone looked at BLE on anything besides A101 yet. So this is invalid, but we should plan to add another supported board that does have BLE at some point and make it work.

closed 2016-12-05 by grgustaf