intel/zephyr.js #298: Approach to Zephyr version upgrade

pfalcon, 2016-10-18:
I see this issue is listed in https://github.com/01org/zephyr.js/issues/125, and open this ticket to discuss specific approach/plan, and to make sure there's no unneeded duplication of effort.

poussa, 2016-10-18:
What we need to do is to update the `deps/repos.txt` zephyr commit id to latest and eventually stick it to 1.6 release and they there for a while. We need to ensure we do all the changes so this repo builds and works on all supported platforms. If there are regressions they need to be documented via issues.

poussa, 2016-10-18:
I tried this earlier today here: https://github.com/poussa/zephyr.js/tree/repo-update

Some of the simple samples work (Button.js, Blink.js) on A101 but e.g Bluetooth, I2C, Groove LCD did not. 

@pfalcon Feel free to continue the work if you are interested.

pfalcon, 2016-10-18:
So, Z.js is currently on the latest Zephyr release, 1.5.0. However, one issue with 1.5.0 is that its IP stack (based on patched uIP) is pretty broken and doesn't really work in a general case for TCP. I spent extensive time trying to fix the most blocking issues to produce a demo of BSD socket like API implementation for another scripting language, and some of those patches are now in the master (to be 1.6.0), while some are too adhoc to be merged, but altogether with them, it all in "barely works" condition.

Instead, there's a "native IP stack" in "net" branch of Zephyr (rebased on master regularly). It seems to work well enough for UDP, and initial patch for TCP was posted. AFAIK, the plan is to include native stack in 1.6.0, alongside of uIP-based stack, and then remove the latter in following releases.

Bottom line: somebody interested in IP-based networking would likely be interested in trying newer Zephyr version than 1.5.0, and overall, trying to follow master makes sense, to avoid extra integration efforts later.

pfalcon, 2016-10-18:
> I tried this earlier today here: https://github.com/poussa/zephyr.js/tree/repo-update
> @pfalcon Feel free to continue the work if you are interested.

Right, so the idea of this ticket is to advocate that trying to follow master (or net branch) makes sense, and then see if somebody already works on that, to avoid duplicating effort. I'll have a look at your branch, and yes, I'd be interested to continue that work. Thanks.

jprestwo, 2016-10-19:
I have this PR that has updated both JerryScript and Zephyr to master as of today (10/19/16)

https://github.com/01org/zephyr.js/pull/304

pfalcon, 2016-10-31:
Ok, https://github.com/01org/zephyr.js/pull/304 which was submitted the same or next day after this RFC, and now merged, after all-hands testing and bug-busting work, answers it pretty well: there's desire to follow upstream version closely enough in devel branch. I assume contributors are welcome to raise future cases where an upgrade would be helpful. Closing then.

closed 2016-10-31 by pfalcon