intel/zephyr.js #301: Linux build is broken

poussa, 2016-10-19:
Tried to build `make linux` as it was failing with @gabrielschulhof 

Both master and devel have the same issue.

I get a different error message on Ubuntu 16.04

````
gcc -o jslinux -flto src/main.o src/zjs_buffer.o src/zjs_callbacks.o src/zjs_event.o src/zjs_linux_time.o src/zjs_modules.o src/zjs_script.o src/zjs_script_gen.o src/zjs_timers.o src/zjs_util.o -L/home/spoussa/src/iot/jerryscript/build/lib/ -ljerry-core -lm -Isrc/ -I/home/spoussa/src/iot/jerryscript/jerry-core -DZJS_LINUX_BUILD -DBUILD_MODULE_EVENTS -std=gnu99 -Wpointer-sign
lto1: fatal error: bytecode stream generated with LTO version 3.0 instead of the expected 4.1
compilation terminated.
lto-wrapper: fatal error: gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
Makefile.linux:44: recipe for target 'linux' failed
make[1]: *** [linux] Error 1
make[1]: Leaving directory '/home/spoussa/src/iot/zephyr.js'
Makefile:214: recipe for target 'linux' failed
make: *** [linux] Error 2
````

````
spoussa:zephyr.js$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.2) 5.4.0 20160609
```

jprestwo, 2016-10-19:
@poussa, @gabrielschulhof,

I just did a fresh clone of the repo and I am not seeing this error. I am on Ubuntu 16.04. I have never seen this error message but it could be because of a stale library that was compiled with a different version of gcc/ld. I assume you have tried a clean build?

poussa, 2016-10-19:
I re-try. I was in hurry and may have missed something.

brianjjones, 2016-10-19:
I'll try as well

brianjjones, 2016-10-19:
Both master and devel work for me on 16.04, and I'm using the same gcc.  Let me know if there's any other settings you need me to check to see if we can get this sorted out.

poussa, 2016-10-21:
My setup issue. I had JERRY_BASE pointing to a repo which was build with Ubuntu 15.04. I updated  to 16.04 which has newer compiler. Seems that make clean won't clear the JrS libs.

````
$ make clean
$ find . -name '*.a'
./deps/iotivity-constrained/deps/tinydtls/libtinydtls.a
./deps/jerryscript/build/lib/libjerry-libc.a
./deps/jerryscript/build/lib/libjerry-core.a
./deps/jerryscript/build/lib/libjerry-libm.a
```

closed 2016-10-21 by poussa