intel/zephyr.js #310: [Timer] Timer alloc failed

cuiyanx, 2016-10-20:
The timer has limited number when be created, and be shared by setTimeout(), setInterval() and GPIOPin.openAsync(). The maximum of timers is 4 at now. 

Error Log:
    native_set_interval_handler: timer alloc failed
    JerryScript: cannot run javascript

Sometimes it well be stucked, look like stoped. but it well be running and ending after added any functions above setInterval() function.

grgustaf, 2016-11-04:
This works if you build with MALLOC=heap. @jprestwo maybe look at why this is failing with pools and see if the values should possibly be tuned to allow more than 4 timers. (I didn't verify that's the max.)

cuiyanx, 2016-11-07:
Although they change timer using MALLOC, but it still do not fix issue 310, and using MALLOC can help to verify our tests result, so keep eyes on 310, i think they should have other fix patch for it.

jprestwo, 2016-11-10:
So the issue is that there are not enough pool blocks for how big timers are. Timer allocations are 60 bytes, and we have just 10 64 byte slots, meaning we can have at most 10 timers and that is if there is nothing else taking up those 64 byte slots (like the previously mentioned GPIOPin.openAsync or other module allocations). This is just going to require some tuning, but we are going to have to make a decision on the max number of timers we allow. Obviously its quite low now.

The best solution would be to dynamically create a pool configuration based on the amount of RAM available after all the modules are compiled in. This is a bit more difficult but I think its worth looking into.

xiaolongx, 2016-11-15:
This issue cannot reproduce on both Arduino 101 and FRDM_K64F with commit 2ff271b, but another issue found in FRDM_K64F(#447)

grgustaf, 2016-12-06:
Apparently the immediate issue is fixed; we'll still want to revisit the effective limit on timers and how we could improve that or make it tunable by the developer.

closed 2016-12-06 by grgustaf