intel/zephyr.js #311: [GPIOPins] Fail to close GPIOPins

cuiyanx, 2016-10-20:
After closed the GPIO pin, I can read and write this pin. It is different to description of GPIOPin.close function.

GPIOPin.close
void close();
Free up resources associated with the pin. The onchange function for this pin will no longer be called, and the object should not be used for reading and writing anymore.

grgustaf, 2016-10-25:
Technically, it doesn't say that won't work. It says you *shouldn't* do that. (I.e. it might not work in the future.) But yeah, probably we should force it to not work so people don't rely on this behavior.

cuiyanx, 2016-10-27:
So, I think it is better method of forcing it to not work.

grgustaf, 2016-12-23:
FWIW I see onchange does get disabled, but you're right, read/write don't.

grgustaf, 2016-12-23:
This should be fixed by #577.

closed 2016-12-26 by cuiyanx