intel/zephyr.js #315: [ashell] "to cancel" is missing in the usage information for 'load'

xiaolongx, 2016-10-21:
Following the example in [ashell.md](https://github.com/01org/zephyr.js/blob/devel/docs/ashell.md), "to cancel" is missing when loading a file in ashell output.

```
Example of raw text input
acm> load test.js

Saving to 'test.js' 
Ready for JavaScript. 
        Ctrl+Z to finish transfer.
        Ctrl+X or Ctrl+C to cancel.
RAW> 
```

Actual Output:
```
acm> load test.js

Saving to 'test.js'
Ready for JavaScript.
        Ctrl+Z or Ctrl+D to finish transfer.
        Ctrl+X or Ctrl+C
RAW>
```

grgustaf, 2016-12-06:
Please update documentation.

grgustaf, 2017-03-24:
@brianjjones this should be worth doing for 0.3, bumped to P1.

brianjjones, 2017-03-24:
This will be fixed by https://github.com/01org/zephyr.js/pull/891

qiaojingx, 2017-03-28:
Verified with commit 694cf15 in master.
Still got inconsistent output when using load and eval command.
[image]
[image]

[image]
[image]

brianjjones, 2017-03-28:
https://github.com/01org/zephyr.js/pull/926 should hopefully do the trick.

qiaojingx, 2017-03-29:
Verified with commit 1c67773. This issue is not reproducible. Closed

xiaolongx, 2017-03-30:
Closed according to above comment.

closed 2017-03-30 by xiaolongx