intel/zephyr.js #325: [RFC] Adding basic time function support

pfalcon, 2016-10-25:
I see that currently Zephyr.js doesn't seem to have any means to measure time intervals. I'd like to add basic support for this, for benchmarking, and well, to measure time intervals. At this time, no talking about datetime (real time, wall clock time).

Quick survey on how to do it in JavaScript brought:

1. Standard down to ECMAScript Date object. The issue here is that this is pretty overloaded object, used for datetime functionality too. `Date.now()` (static method) is speced as "returns the number of milliseconds elapsed since 1 January 1970 00:00:00 UTC."
2. Node.js has `process.hrtime()` returning Array of [time_since_epoch_in_seconds, nanoseconds]
3. Draft "High Resolution Time" spec from W3C: https://www.w3.org/TR/hr-time/ , https://developer.mozilla.org/en-US/docs/Web/API/Performance/now . This would appear as the most relevant to the topic of benchmarking and what we can achieve out of the box with Zephyr - it doesn't depend on some absolute epoch, but on a relative time.

My personal preference would be go Date.now() way, as it's more widely known. It may be also first step of supporting other Date features (absolute time). Note that Date is implemented (as in: available) in JerryScript, so implementing Zephyr's version there would allow reuse by other JrS-based project. There's a risk too, if JrS maintainers won't agree that for the lack of better alternative, it's acceptable to use a relative time (since boot of a board), instead of absolute Unix Epoch time. Then, Date implementation would need to be migrated to Zephyr.js.

pfalcon, 2016-10-25:
> To the extent we've thought about it at all, the vague plan would have been to try to pull in JrS version at some point, but maybe working on some config options to pull in just what we really need or the actual app uses and not everything. (We have a similar issue for Math etc.)

I see. Well, I'd say that may be spreading too thin. It's one thing to include/omit whole modules, and rather different matter to exclude/omit individual objects or their methods. Non-trivial program analysis may be required for the latter vs just grepping for "require(.+)". Anyway, indeed, to optimize something, one first need to have it implemented ;-).

> (Apologies that we've got a bit of a patch logjam this week as we work on CI and such; we appreciate your contributions and will try to get them merged ASAP.)

No problem, I see that there's a lot of important activity happens.

> I don't think I understand the relative time issue you refer to at the end there.

Well, perhaps I don't use the right terminology (but there's no consistent terminology accepted in IT). There're 2 time sources: # 1: ones with a well-known reference ("seconds since 1 Jan of ..."), and # 2 ones  without (just a free-running counter started at "random" (not absolutely referenced) moment). 1st kind can be used for a) getting an absolute time of this moment ("now is hh:mm:ss of YY-MM-DD"); b) to measure time periods (differences).

# 2 type can be used only for the purpose b). Users of desktops got used to clocksource of type # 1 (most APIs promise you absolute time), while in embedded we usually have # 2. In this regard, W3C's hr-time is more of type # 2 (though it still has a field to reference its relative time to absolute reference, reading formal spec). Bottom line: it's hard to port "desktop" stuff (like JavaScript) - there will be differences which will confuse users (and which users should account for).

pfalcon, 2016-10-26:
Enabling Date builtin in JerryScript costs 17780 bytes of ARM Thumb2 code. (123224 .bin size for qemu_cortex_m3 vs 141004 with the feature enabled). I'm going to flush changes needed to enabel it upstream, but @grgustaf , @poussa let me know if you're still interested in such approach, or rather would prefer an alternative (implement Date with just .now() in zephyr.js, implement performance.now() in zephyr.js; I'd say the latter looks more and more promising if you're concerned with code size).

pfalcon, 2016-10-26:
The Date enablement posted upstream as https://github.com/Samsung/jerryscript/pull/1411

poussa, 2016-10-27:
My vote goes to implement the W3C performance API in here. It goes with our goal to bridge the API gap between the browser and embedded JS environments and cost less memory.

pfalcon, 2016-10-27:
@poussa : Sounds good. As this is an RFC, I'll mention yet another option: just make an adhoc "zephyr" module, and add various functions there, and later see if there's "official" (in some sense of that word) JS API for it, and what it would cost to implement it properly. My choice would be, whenever possible, to try to look for a suitable JS API right away, but a module like above may need to be created anyway, to host various helper/debugging/introspection functions.

poussa, 2016-10-27:
@pfalcon I like the idea of experimental module - not sure if we want to call if zephyr though.

That could the the API cooking place when we are not sure how the API should look like.

Like to get a comment from our API spec guys, @zolkis & @kenchris !

kenchris, 2016-10-28:
The performance API looks a bit browser specific but we might be able to model something after it (it seems that various types inherit from a more generic one). I found a primer here:

https://w3c.github.io/perf-timing-primer/

pfalcon, 2016-11-01:
> https://www.w3.org/TR/hr-time/ is probably the most simple we can get

I find the most practical description being a dev-mozilla link given in the original description -  . https://developer.mozilla.org/en-US/docs/Web/API/Performance/now . It shows that it is essentially just:

~~~
t = performance.now();
~~~

That's it. Yes, the formal spec from W3C has more stuff, but it would be rarely used, and per our discussion in https://github.com/01org/zephyr.js/pull/344#issuecomment-257576388 , would be out of scope of Z.js for the time being. With that in mind, proposed implementation is in https://github.com/01org/zephyr.js/pull/346

pfalcon, 2016-11-10:
Implemented in https://github.com/01org/zephyr.js/pull/362

closed 2016-11-10 by pfalcon