intel/zephyr.js #339: RFC: What should we do for versioning?

grgustaf, 2016-10-28:
We need to start having versions, as @poussa has already asked us to do. If someone else doesn't have a better idea I think we should call our Zephyr 1.5 work 0.1, and call our new Zephyr 1.6 work 0.2. Then we'll set some criteria regarding relatively complete APIs, stability, and user-friendliness to call it 1.0.

Where does this version need to sit? I guess we should be able to check the version at build time and at runtime with an API. How do we identify final "golden" version vs. pre-release. And then what do we call it if we have to fix the official release?

Comments and wisdom welcome!

To answer my own questions with a suggestion, maybe as soon as we start working on zephyr 1.6 (well already happened), we call our version e.g. 0.2-pre, then when it's golden we call it 0.2, and if we need to fix it we call that 0.2.1-pre until it's stable and then 0.2.1, etc? I guess we can just put this string in a version file at the toplevel or something and then generate C code to make it accessible to an API.

baleboy, 2016-10-28:
+1 on the version names, -pre etc.

grgustaf, 2016-12-05:
So what we've settled on for now is that we have a version file in the root dir that contains the name of the version. On the devel tree this is just 'devel' although maybe it would be useful to be more specific. When we branch for a specific version we call it e.g. 0.1-pre. Then if we add milestones they would be e.g. 0.1-alpha, 0.1-beta, 0.1-rc1, etc up to 0.1 when we decide it's done. For 0.1 itself we did not have a lot of process and jumped straight to 0.1 after QA and a team decision.

closed 2016-12-05 by grgustaf

pfalcon, 2016-12-05:
> So what we've settled on for now is that we have a version file in the root dir that contains the name of the version. On the devel tree this is just 'devel' although maybe it would be useful to be more specific. When we branch for a specific version we call it e.g. 0.1-pre.

Just some comment: a common way to find out "which version is this" is to use `git describe`. And here's the current output:
~~~
$ git describe --tags
stable-111-ga162db1
~~~

Not very helpful. So, it's sad that there's no "v0.1" tag on master, as if the master never reached v0.1. But having said that, it's one step from saying "can we release from master, and when it's in a release freeze, do development on devel branch?". But that's how it used to be, and was switched to the current way instead. Well, then I don't say anything except that it's sad that the git describe output is like above ;-).

grgustaf, 2016-12-07:
OK, another thing for me to look into. I hadn't ever used that command before. :)

In this model, master branch never *did* have 0.1 in it, because we branched off in order to finish up 0.1. I guess what I like about that is that big new development work always goes onto master and never has to slow down for releases. There is no rebasing of the big new features landing to get them back onto the tree after a release. But I don't have the terminology to describe these things succinctly and that's good evidence that I don't fully understand the alternatives I'm implicitly deciding between. So I'm sure this could use more thought, and I'm not confident I've chosen the best model for our project.

grgustaf, 2016-12-07:
Hmm, the only way `git-describe --tags` is going to give a more satisfying answer is if we abandon the idea of a "stable" tag (which really we have not been updating properly anyway). Whenever we add a stable tag, that's going to be what shows up as most recent tag. So I'm thinking that command really is not that great for the purpose of identifying a random commit. I did add some tags 0.1-devel and 0.2-devel that at least clarify on the master branch when we started working toward each release. So now the following command will be of some use for that:

```bash
git describe --tags --match "*devel"
```

As far as the branching strategy, I read a bit about [gitflow](http://nvie.com/posts/a-successful-git-branching-model/) and maybe that's more what you're used to. Maybe that's what we were trying to reinvent when we had a 'devel' branch. But we diverged in that we were insisting that devel merge cleanly onto master, which we finally realized was lunacy.

Anyway, as I read through some of that I don't think there's really anything wrong with our model. It's just a bit different. Basically, our master is their develop branch. The release branches are basically the same concept. We just skip the step of merging things back into a single place. I admit that has a satisfying ring to it in that you get to see each release as being directly derived from the last one. But I don't think there's really any huge value we're missing by not having that. Also, something like that could make more sense after 1.0 when you're really in "production". The benefits are very nebulous right now when we really just want people to try out the latest development code.

So let's leave the apple cart alone for the time being.

pfalcon, 2016-12-07:
@grgustaf : Thanks for looking into this. I definitely didn't call to change anything, just shared a non-obvious case where there's a difference between 2 approaches. Otherwise, there're both viable and both used, even in a nearby projects. E.g. Zephyr uses "release from master" approach, while JerryScript uses "release branch" approach. One isn't better (or worse) than another, except ... for git describe. And one of course won't pay attention to it until has a need for it. I decided to use it when setting up CI on our side, it really helps to include source revisions in binary names to be able to track something across 15 always-moving builds: http://snapshots.linaro.org/components/kernel/aeolus/zephyr/jerryscript/frdm_k64f/12/ . As you can see, JrS has it confusing too: tag says it's 1.0-RC2 + something, but of course it's 1.0 + something.

> I did add some tags 0.1-devel and 0.2-devel

That's already quite helpful, thanks!