intel/zephyr.js #341: How to run tests?

pfalcon, 2016-10-31:
Please let me play "casual user" here (I'm going to dig a bit deeper after this report, but I'm afraid one can get only so far with this matter without docs). So, I see `tests/` dir in the tree, I see commit traffic around tests. However, there's no info of them in the README beyond just mentioning `tests/` dir. Ok, I try to run them. "make test", "make check", "make tests" don't work. I decide to run them manually, and assume that the most obvious way to do that is with Linux version, as something which everyone has, though seeing test-aio.js puts me off a bit. Anyway:

~~~
$ ./jslinux tests/test-buffer.js 

JerryScript: cannot run javascript
~~~

No error, no nothing, just cannot run.

~~~
$ ./jslinux tests/test-timers.js 

PASS - clearInterval: intervalID does not exist
PASS - clearTimeout: timeoutID does not exist
PASS - setTimeout: set delay
PASS - setTimeout: set delay and optional arg
PASS - setInterval: set delay
PASS - clearInterval: intervalID
Segmentation fault (core dumped)
~~~

Note the segmentation fault at the end.

pfalcon, 2016-10-31:
I tried to build a test image for frdm_k64f with:

~~~
make BOARD=frdm_k64f JS=tests/test-timers.js
~~~

Result:

~~~
PASS - clearInterval: intervalID does not exist
native_set_interval_handler: timer alloc failed
JerryScript: cannot run javascript
~~~

haoxli, 2016-11-01:
They can run as well as `samples/`:
```
make JS=tests/test-timers.js
```
Unfortunately `test-timers.js` is blocked due to #310, please try other tests, thanks.

baleboy, 2016-11-01:
I think @pfalcon 's comment is valid, a make 'test' target that translates to the test command above would be a simple way to make things easier.

grgustaf, 2016-12-05:
I've assigned to @jprestwo to see what we can do here. You're right, the linux build has only very partial support but we have failed to document what it is useful for and what the plans are.

We should think of some way to identify what tests are expected to work with jslinux. I'm surprised though that nothing I've tried in tests/ seems to work expect for test-eval.js.

grgustaf, 2016-12-05:
So some possible actions:
- Add some documentation to main README and a tests/README.
- Fix jslinux or tests so that more work with it.
- Perhaps create a linuxtest make target that would run everything expected to work, or something. This also seems it would depend on #330. (In turn, once we had that we could add these linux tests to Travis.)

jprestwo, 2017-01-20:
@grgustaf:

I have a PR with a complete list of supported modules on Linux/Zephyr. This should clear up any unclear areas. This doesn't include any actual new functionality so maybe we can leave this open until we have some time to write a complete "unit test" for Linux which tests every supported module.

jprestwo, 2017-03-02:
@grgustaf, can this be closed? Docs are on master explaining how to build/run the linux target.

jprestwo, 2017-03-22:
@grgustaf, @pfalcon,

Can one of you close this, docs should be up to date.

pfalcon, 2017-03-22:
Thanks, closing.

pfalcon, 2017-03-24:
@jprestwo :

> Can one of you close this, docs should be up to date.

I'm not sure which docs you meant. I re-read README in full, and the only mentioning of tests I found was btw-style note of:

> Note: sysvbanner is just used to identify tests as they run in the trlite test build script, so it could be left out.

I thought maybe "make help" now has something, but nope.

So, for the interest of the project, please let me reopen this ;-).

jprestwo, 2017-03-24:
@pfalcon, 
Sorry, I guess I confused the intention of this issue with my own thinking. I can get into thinking things are "obvious" but really for a new user that would not be the case. I wrote a section in the README about running ZJS on Linux, as well as a chart of what modules worked on Zephyr and on Linux. But if it needs to be more clear on exactly which tests (i.e. the files specifically) work on Linux and which work on Zephyr we can write up another README for the test directory, which I do think is a good idea. Is that more of what you were getting at? Or would it be better as comments in the specific test files that say which system(s) they run on?

pfalcon, 2017-03-25:
> Is that more of what you were getting at?

Nope. There should be a streamlined, fully automated testsuite (in addition to any other testsuites, like half-automated or fully manual). By "streamlined" I mean running quick, and being always at developers' fingertips, so people actually ran it (frequently).

> The tests themselves are just JS applications so you run them as you would a sample.

That exactly the current situation and the problem of Z.js - there're "tests", as in, unordered, random collection of scripts, instead of the automated regression testsuite, which is a software development tool which prevents regressions on-spot (usually right at the developers' machines). It consists of hundreds tests and thousands of testcases, so there can't be talking about running them one by one manually.

This local testsuite should be the basis of CI, whereas as far as I can you guys have it the other way around - you have Travis, then some (undocumented!) "travis lite" script which tries to run it locally as if it was a full CI server. Oops, complicated, I already terrified as a contributing developer (partly because I know that it won't work for me how you intend it to - simply because with all its git clone's and gazillion of other git commands it's too complicated to do the right thing).

YMMV and feel free to take it with a grain of salt and wait while other contributors with their cry-of-soul's appear. I'm writing this under impression of https://github.com/01org/zephyr.js/pull/842#issuecomment-289202305 , where even project maintainers make changes from correct to incorrect and there's nothing to detect that until too late.

jprestwo, 2017-03-27:
@pfalcon,

Ok I see what you mean. I did just push a PR to run a limited set of tests on Linux for each build, but thats not really testing full functionality. I agree we need "something" but its rather difficult due to the fact that most of our tests only run on the Arduino 101. Which requires you to flash (press reset, wait, press again) to run the next JS file. We could try consolidating every 101 test into a single file, but size constraints may come into play. Its definitely worth looking into.

pfalcon, 2017-03-27:
@jprestwo :

Thanks, much appreciated. And we're getting there - slowly, let's continue our way down there. So, PR you mention is https://github.com/01org/zephyr.js/pull/895, and so it updates scripts/trlite, which is, as I already guessed, is you testsuite runner. So, It ry to run it myself:

~~~
$ ./scripts/trlite 
Building ZJS tree: /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr.js
Cloning git tree...
^C
~~~

As you can see, it tries to git clone something, and my reaction is immediate - I break it. That's because I know an invariant of all times and places: **To run a testsuite, there's no need to perform any git operations.** No matter what great reasons you have to do it like above, but I'm sorry to put it like that - these reasons are wrong. To run a testsuite, you just need to enumerate available testcases, and then execute them in order, checking for failures. No more and no less than that, no git operations involved. Just like your patch in https://github.com/01org/zephyr.js/pull/895/files does.

Consequently, can you please consider:

1. Either making a separate testsuite runner devoid of git operations. I extracted your patch into a separate script file in https://github.com/01org/zephyr.js/pull/905 to show how that would look like. Granted, this might be not the ideal solution, because it violates DRY principle. But it's bright side is that we'll be able to keep it a quick-running subset of tests (which is a prerequisite for people actually running it themselves).
2. Change trlite script to not perform any git (and other unrelated networking i/o etc.) operations, at least by default. What that script needs to test is right there - my local tree, with whatever changes I have in it.

Thanks for your consideration.

grgustaf, 2017-03-27:
One step of progress is in #907, which adds "make check" and "make quickcheck", which call trlite w/ new git-free option.

@pfalcon, we have tests in there that use "git diff" and "git check" though, is that okay with you?

Note to self: come back and actually read the whole thread here.

pfalcon, 2017-03-28:
> we have tests in there that use "git diff" and "git check" though, is that okay with you?

I guess ok, at least they're quick and hopefully don't modify anything on local side. I'll need to look in more detail into them, and will raise an issue if anything ;-).

> Note to self: come back and actually read the whole thread here.

I'd say, that won't be too good use of time, as this ticket accumulated a lot of content over months, some no longer pertinent. The main topic of this ticket is in its title: "How to run tests?", and https://github.com/01org/zephyr.js/pull/907 now answers that well. So, let me close this finally and raise any new issues/suggestion separately.

Thanks for following thru this, guys!

closed 2017-03-28 by pfalcon