intel/zephyr.js #342: Support for multiple .js source files?

pfalcon, 2016-10-31:
(Sorry for possible stream of tickets, I tried to "ration" "generic questions" vs specific RFCs/patches, be we (Linaro) now talk on our side on having a release for Zephyr-related things, so I decided to dump whatever I had in mind, to have better idea what can planned, and for when).

So far, the biggest limitation of Zephyr.js appears to be allowing to using just a single JavaScript source file. I hope everyone would agree that's a limitation, as any realistic app would have more modular structure and would want to reuse previously written modules. And well, that's the point - it's not about "supporting multiple JS files", it's about supporting modules written in JS, not just "native" modules written C as it's possible now. And allow to construct and application from many such JS modules, not just from many C-coded modules as it is now.

This of course leads to question which module approach to use, as there's no single generally accepted module system for JS. Well, perhaps one from EcmaScript6 would become one, but it's not currently, with module systems predating it in wider use.

I'd like to hear Zephyr.js' architects opinion on this matter. @poussa , @kenchris.

jprestwo, 2016-10-31:
I believe this is more of a limitation of JerryScript. We could easily concatenate multiple JS files together and run that, but I don't think that is what you are getting at. We as well would like the ability to write JS modules in JavaScript that can be required/loaded from a different/main application, but, last I checked, this is not possible with JerryScript. I could be wrong, but its definitely something we can and should revisit now.

pfalcon, 2016-10-31:
> I believe this is more of a limitation of JerryScript.

Right, JrS is ECMAScript5 implementation, so doesn't have any builtin modules support, and it would need to be implemented on top of it. IoT.js definitely supports JS modules. To remind, they went as far as setting up their own npm-like repository for modules (later they shut it down, apparently because it's one thing to do such a cute hack, and another - to maintain it over time and support its users).

> We could easily concatenate multiple JS files together and run that, but I don't think that is what you are getting at.

Right, I'm talking about being able to do `var mymod = require("mymod")`, which would import `mymod.js` as a "module" (whatever definition of module will be for Z.js).

It's definitely doable - the first step is to switch from `const char *script_gen = <code>` to something akin to:

~~~
var scripts =  { 
  "main.js": "<code>",
  "mymod.js": "<code>"
}
~~~

Then execution would start with looking up code for "main.js" (by-convention name of app's main script, or can be made configurable) in `scripts` dictionary above and then run it just like before. But "require()" implementation now would look into `scripts` too, not just into similar dictionary for builtin C-coded modules. There're lot of details beyond that: specific module paradigm to follow (well, the main question is how to export things from a module, the rest is syntactic sugar), namespacing for execution of unrelated modules, etc. But the general scheme should be like above (in a sense that above should be pretty viable scheme for starters).

> but its definitely something we can and should revisit now.

As I mentioned, I didn't mean to steal people's attention with this ticket, as there's more ground work to do for now. But if interested parties could think about this and comment e.g. over next month, I guess it would help all parties interested in Z.js to have better understanding of what directions it may evolve.

Thanks!

poussa, 2016-10-31:
We would definitely need to support multi file JS applications so thanks @pfalcon for initiating the discussion. 

In my books we have two options: a) Node.js style CommonJS (CJS) modules or b) ES6 modules.

In the ideal world, I would like to go with ES6 modules since they are 1) standard, 2) seem to be future proof and 3) even the Node.JS is going to that direction. Node.JS has huge legacy of CJS modules so their energy is mostly spend on figuring out the migration path.

As it was mention, JrS does not support ES6. AFAIK, it is not in their 2017 plans either. I was looking their (JrS) issues but there was nothing about ES6 modules. We should start the discussion there as well. The other option (for ES6 modules) would be what we did with Promises. That is, just implement them in the ZJS. Or maybe the combination of the two. First, implement partial support in ZJS for ES6 modules and once the feature lands in JrS, adopt that.

pfalcon, 2016-11-10:
https://github.com/01org/zephyr.js/pull/400 is a progress in this direction.

grgustaf, 2016-12-05:
@brianjjones please review and update with the latest status and comment on whether anything additional should be aimed at 0.2 release.

brianjjones, 2016-12-05:
Master currently contains support for the Node.js style of including modules.  You can take a look at samples/I2CBMP280.js for an example of how to use this feature.  Any JavaScript modules should be placed in the 'modules' folder so they can be detected.  As for the ES6 way, I have a simple script that can support that style of inclusion, but it was decided we'd leave it out for now as to not complicate things.  Hopefully in the future JerryScript will have native ES6 module support, but it hasn't been planned yet.

brianjjones, 2016-12-10:
@poussa @pfalcon  Does this https://github.com/01org/zephyr.js/pull/423 cover this issue?  If not what else needs to be added before we consider this complete?

pfalcon, 2016-12-10:
Shame on me, I still didn't have a chance to play with it (we're preparing first Linaro IoT release on our side, so mostly working on "boring" issues like CI setup in the meantime). But I'm astonished with the speed this issue was acted on, thanks, and let me just close it for now.

closed 2016-12-10 by pfalcon