intel/zephyr.js #343: console.log() behavior is not compliant

pfalcon, 2016-10-31:
To reproduce, modify samples/HelloWorld.js (do we get an interactive mode?) to contain:

~~~
console.log(1)
console.log('foo', 'Hello, ZJS world!')
~~~

Run:
~~~
$ ./jslinux samples/HelloWorld.js 

foo
~~~

Issue: numeric argument is not printed. With multiple arguments, only 1st is silently printed.

For comparison:

~~~
$ nodejs samples/HelloWorld.js 
1
foo Hello, ZJS world!
~~~

Browser interactive console:
~~~
console.log(1)
console.log('foo', 'Hello, ZJS world!')
1
foo Hello, ZJS world!
undefined
~~~

kenchris, 2016-10-31:
https://console.spec.whatwg.org/

pfalcon, 2016-10-31:
I also tried the HelloWorld.js as modified per above on frdm_k64f, with the same result:

~~~
$ picocom -b115200 /dev/ttyACM0 
...
Terminal ready

foo
~~~

In other words, currently Z.js doesn't have means to print numbers (which is hard to believe, so maybe regression after recent JrS/Z upgrade?).

Note that upstream JrS with its print() feels well in this regard (Z.js's print() doesn't print numbers still):

~~~
$ picocom -b115200 /dev/ttyACM0 
...
Terminal ready

JerryScript build: Oct 31 2016 17:31:56
JerryScript API 1.0
Zephyr version 1.5.99
js> print(1)
1
undefined
js> print("foo", "bar")
foo bar
undefined
js> 
~~~

jprestwo, 2016-10-31:
I wrote console.log to behave the exact same as print. I did not foresee the event of providing JUST a number like `console.log(1)`, which is definitely an acceptable use case. Good catch, ill get this fixed.

pfalcon, 2016-10-31:
Thanks. I hope you have an idea how to fix it well, because I so far don't have a clear picture. But I'd suggest to consider not trying to have own implementation, but reuse JrS' print() implementation - it already handles multiple args, and Z.js carries its code anyway. Indeed, a workaround I use so far is:

~~~
-    zjs_obj_add_function(global_obj, native_print_handler, "print");
+//    zjs_obj_add_function(global_obj, native_print_handler, "print");
~~~

This stops overriding JrS's own print().

How to implement console.log() in terms of JrS's print() is another question. I checked that their implementation is static, and calling them to un-static it for reuse wouldn't sound completely ungrounded for me.

jprestwo, 2016-10-31:
Unfortunatly, after doing a JerryScript and/or Zephyr upgrade, the default print function was broken which is why we had to override it. There are a few options we have to get this fixed, and in the worst case we just parse the args ourselves and print them. Luckily there are just numbers, strings, and arrays if we decide to go with that capability. I assume this will be low priority because the work around is simple in JS. Instead of doing:
```
console.log(1);
```
you can do:
```
console.log(1 + '');
```
which will automatically turn the whole argument into a string. Multiple arguments won't work though.

haoxli, 2016-11-01:
Same with issue #319

pfalcon, 2016-11-01:
> Unfortunatly, after doing a JerryScript and/or Zephyr upgrade, the default print function was broken which is why we had to override it.

As I mention above, using it works fine for me now. Well, I don't want to get stuck on this matter - you guys know better how it should be done, just wanted to leave a more complete discussion is the history, if this issue will be revisited in the future.

pfalcon, 2016-11-01:
@haoxli 

> Same with issue #319

Sorry for this duplicate. Well, as this ticket has more discussion now, perhaps it makes sense to move discussion from #319 here too.

xiaolongx, 2016-11-07:
#344 fix part of console issues, but there are still two issues unresoved:

1. Cannot output boolean value: [DigitalReadSerial.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/basics/DigitalReadSerial.js)
2. Cannot connect String and variable using ',' 
```
var str="b";
console.log("a", str);
```

jprestwo, 2016-11-10:
This should now be fixed with: https://github.com/01org/zephyr.js/pull/413

closed 2016-11-10 by pfalcon

xiaolongx, 2016-11-24:
This issue still reproduce on zjs-0.1 branch (commit dcb06fb)