intel/zephyr.js #349: BLE notification bug on subscribe/unsubscribe

jimmy-huang, 2016-11-02:
There's a bug currently in BLE that it immediately start sending notification to the client as soon as the client is connected, since before we couldn't detect which service the client subscribed to due to Zephyr bluetooth driver limitation, now Zephyr upstream added an extra param in their ccc_cfg_changed() callback, we should fix the BLE to monitor subscription/unsubscription on notifcation from the client, and only send update notification if they've subscribed.

grgustaf, 2016-12-05:
@jimmy-huang, is this done?

jimmy-huang, 2016-12-09:
submitted a fix here https://github.com/01org/zephyr.js/pull/528

xiaolongx, 2017-01-12:
This issue should be fixed on Arduino 101 with commit 2345f96.

closed 2017-01-18 by grgustaf