intel/zephyr.js #359: [PWM] Error thrown when set pulsewidth before period

cuiyanx, 2016-11-03:
```
var pinA = pwm.open({ channel: pins.IO3 });
pinA.setPeriod(1000);
pinA.setPulseWidth(300);
```

It work well.

```
pinA.setPulseWidth(300);
pinA.setPeriod(1000);
```

**error message:**

`zjs_pwm_set: pulseWidth was greater than period`

grgustaf, 2016-12-05:
My conclusion looking at the Zephyr API was it's not good to set these things separately because you're in an invalid state in between. They agreed and changed their API based on my suggestions. So hopefully we can follow suit. @zolkis may be interested, I am afraid I'm not digging in at the moment to see what it's like in the new APIs currently.

grgustaf, 2017-03-29:
This is addressed by #928. Now the APIs require you to set period and pulseWidth at the same time. If pulseWidth is greater than period, they throw an error immediately.

qiaojingx, 2017-03-29:
Verified with commit  1c67773 + #928. This issue is not reproducible.
Will close it after patch merged.

PR #928 (2017-03-30): [pwm] Require PWM period/pulseWidth to be set at the same time
This is a wise API choice because you can go through invalid states
by setting the wrong one first, where the pulse width is bigger than
the period. I convinced Zephyr itself to make this change, but we
hadn't gotten around to reflecting this in our own API.

Fixes #359.
commit d56bac22 (2017-03-30): [pwm] Require PWM period/pulseWidth to be set at the same time (#928)

This is a wise API choice because you can go through invalid states
by setting the wrong one first, where the pulse width is bigger than
the period. I convinced Zephyr itself to make this change, but we
hadn't gotten around to reflecting this in our own API.

Fixes #359.
closed 2017-03-30 by jimmy-huang