intel/zephyr.js #367: Document release and branching strategy

baleboy, 2016-11-04:
Our release and branching strategy needs to be documented in the wiki so that project contributors and users are aware of it. In short the strategy is:

- Do development on master
- Have quarterly releases aligned with Zephyr releases
- Branch for each quarterly release
- Base each stable release on a stable version of Jerryscript and Zephyr

baleboy, 2016-11-04:
First version here: https://github.com/01org/zephyr.js/wiki/Workflow-and-Releases

@poussa @grgustaf @ibelem

grgustaf, 2016-12-05:
I made some tiny edits. I also added the [Releases](https://github.com/01org/zephyr.js/wiki/Releases) page recently with info on specific releases, and I added a link to Francesco's page on the main wiki page.

So I'm thinking this issue can be closed now.

closed 2016-12-05 by grgustaf