intel/zephyr.js #368: RFC: Making GPIO access board neutral

pfalcon, 2016-11-04:
Currently, Zephyr.js officially supports only 2 boards: arduino_101 and frdm_k64f. Of course, people will want to run it on other boards, and for example I'm already being approached with questions "how to run it on ARM board X?". So far I could answer only "Just try to build it for your board and report any issues upstream". I finally took a look at what would it take to support another board, specifically for GPIO module, and see that the issues are many. Current zjs_gpio.c implementation has wild hard-coding's like:

~~~
#ifdef CONFIG_BOARD_FRDM_K64F
#define GPIO_DEV_COUNT 5
#else
#define GPIO_DEV_COUNT 1
#endif
~~~

So any board but frdm_k64f would get support for just 1 GPIO port. 

Note that support each board one by one is never going to scale. There's only 2 supported now and there're incompatibilities/inconsistencies already: https://github.com/01org/zephyr.js/issues/284 . So, I'd like to propose the following straightforward solution:

* Zephyr.js should be able to support GPIO access for any Zephyr-supported board in a generic manner.
* However, if some party desires to add conveniences for a particular board, like currently done for arduino_101 and frdm_k64f, they should be able to (in particular, all current infrastructure for arduino_101 and frdm_k64f should remain compatible with a proposed additions).

And the way to support generic GPIO access for any Zephyr board is to just use Zephyr's way of identifying GPIO ports and pins, without trying to add any abstractions on top of it. So, Zephyr uses model where there're explicit GPIO ports, identified by strings (effectively device names) and pin numbers within port. This model is of course based on the actual reality of it - (almost?) every MCU vendor splits GPIOs into ports with relative numbers of pins within each port, so GPIO "coordinates" are "two-dimensional".

The specific proposal is:

Update `dictionary GPIOInit` in https://github.com/01org/zephyr.js/blob/master/docs/gpio.md to (additions marked with "+"):

~~~
dictionary GPIOInit {
+   optional string port;
    unsigned long pin;
    boolean activeLow = false;
    string direction = "out";  // in, out
    string edge = "any";       // none, rising, falling, any
    string pull = "none";      // none, up, down
};
~~~

If a `port` is specified, a new model is used (and `pin` is a relative pin number within specified port). Otherwise, the current model is used, where `pin` is a thru-number, encoded in a board-specific way.

poussa, 2016-11-07:
We are moving towards more generic APIs in this area. 

PTAL: https://github.com/01org/iot-js-api/tree/master/api/board

It would be good once you start coding you would adapt that API instead.

Any feedback of the API is welcome. It is just the 1st iteration.

@zolkis is the author.

grgustaf, 2016-11-07:
It's a good point, I think you're right the only way for an unanticipated board to be supported is to pass through raw values to Zephyr. I pointed out to @pfalcon this [new API](https://github.com/01org/iot-js-api/blob/master/api/board/gpio.md) we plan to move to, but I don't know that it addresses the issue either.

(Forgot to send this and got scooped by @poussa :D.)

pfalcon, 2016-11-07:
Thanks, will look thru it.

zolkis, 2016-11-09:
We agree on that. 

In a parallel effort that may eventually be implemented in zjs, namely in the [GPIO API](https://github.com/zolkis/iot-js-api/blob/master/api/board/gpio.md) defined as part of  [iot-js-api](https://github.com/01org/iot-js-api), GPIO pins and ports can be initialized with numbers or strings defined by boards. A given board _implementation_ will own the semantics and valid values. For the apps it's an opaque string as defined in the board doc. The gpio init function will check if the provided value matches a pin or port and provide a suitable object to the app.

daniel-thompson, 2016-11-09:
Just been discussing with @pfalcon on IRC...

I think there is an interesting opportunity to exploit the strings as pins to have a "generic-zephyr" board with pins such as "GPIOA-7" (where once the -<pinno> is stripped off we are left with a string that we can pass to device_get_binding() ).

This doesn't stop zephyr.js also offering board such as "iot-board-arduino101" as well but, on the assumption that the number of zephyr boards is about to grow very fast, it does mean there is no need to constantly chase the upstream and maintain a huge library of lookup tables.

pfalcon, 2016-11-09:
> A given board implementation will own the semantics and valid values.

Thanks for reply, @zolkis. And the quoted phrase is exactly our with @daniel-thompson concern. We would like to make sure that it was possible not a *board* to define semantics and values, but the *OS*. That would be the whole idea why we base on Zephyr RTOS - because it offers generic, standard services to access various hardware devices. So, someone can spend time and add a board support to Zephyr, but afterwards, an application can be written for Zephyr largely abstracted from a particular board, and would work on any board with only *configuration* changes, not code changes to support a new board.

Of course, in our case, the "application" is Zephyr.js, and the talk is about letting it access any GPIO exposed by Zephyr in a way Zephyr exposes it. As @daniel-thompson mentions, that doesn't preclude additional "convenience" layer for particular layers (but the whole idea is to make such layer optional, as writing 100 layers for 100 boards doesn't scale, and I hope we'll have more than 100 boards running in Zephyr not too far from now).

zolkis, 2016-11-09:
I was under the assumption that the board is given, it has its own labels printed on it, and Zephyr + zjs is an add-on. Also, the assumption was that developers will look at the physical board and would expect the board labels to be trackable in source code. That's why the module that implements the board would expose a namespace that reflects the board labels. 

That basically means the API is generic, and accepts whatever is valid for the given board. 
If two boards implement the same name space, then what you want is feasible, but it is not a concern of the API, but the API implementation on both boards.

Your use case seems to be that app code should be compatible across various boards, given they have the same notions of e.g. GPIO pins or ports, and you expect an API that would impose a standardized name space (where such standardization is possible)? So that you'd say give me `GPIO-3` or `gpio(3)` or something like that, and expect the implementation and OS to figure out whether there is a GPIO-3 on that board, whether it's standalone and not part of a port, whether a port of 8 pins or 32 bits etc?  I think that's quite difficult to achieve and it doesn't scale either.

Would you expect to have a board introspection layer to give you the available GPIO pins and ports?

pfalcon, 2016-11-10:
> I was under the assumption that the board is given, it has its own labels printed on it, and the assumption was that developers will look at the physical board and would expect the board labels to be trackable in source code.

That's definitely true. Except that from my perspective, I would call that target group "end users". As a platform engineer, I'm concerned to make board vendors' life easier, and unlock Zephyr's full potential. So, the practical question if Zephyr.js wants to be:

1. An end user product, i.e. ship specific binary firmware for specific set of supported boards (currently 2).
2. Be a "default" JavaScript engine for Zephyr, offering hardware vendors a generic and streamlined platform to build their products on.

Zephyr itself is definitely not an end-user product, but an integration platform, so it's quite natural to see Zephyr.js in the same way. And I think such outlook will be very beneficial for the project (raise adoption levels steeply). Note that in no way that precludes providing "end-user ready" configuration examples within Zephyr.js .

> The API itself is generic, and accepts whatever is valid for the given ("required") implementation: board, or OS.

I would have more comments on proposed API, just need to read it in full and think out my replies.

> I think the concerning sentence could be reformulated like:

>> "The API uses the values and semantics as exposed by the implementation (specific to board, or to the OS). Applications can require the implementation that suits their needs."

> @pfalcon, @daniel-thompson: Would that resolve the concern?

Yes, definitely, that make it clearer on the spec side. I however would like to look at it from more practical side: how do we implement it in Z.js (and I could use that functionality right away).

So, with the outlook of the new API, I'm not going to keep proposing to add separate "port" parameter - methodologically it's right, but practically can be cumbersome. So, does @daniel-thompson's idea of identifying Zephyr-level pins with "two-coordinate strings" (instead of separate port id and pin no) work? I'd say that separating with a dot probably even better, e.g.:

~~~
// A blue LED on frdm_k64f
"GPIO_1.25"
// An LED on recently added 96b_carbon
"GPIOA.15"
~~~

Please pay attention to the last example - some boards don't even follow "GPIO_<number>" naming convention for GPIO ports. (So current Z.js `snprintf(devname, 8, "GPIO_%d", i);` wouldn't work).

daniel-thompson, 2016-11-10:
In addition to reformulating the wording in the formal section I'd also suggest updating the GPIO example code to use strings. The way the current example uses numbers starts the reader thinking from the "incorrect" mental model based on pin numbers that can be hard to transfer to other boards. I think it should still demonstrate a real board (rather than "iot-js-zephyr") but in my experience very few boards put pin numbers on the silk screen. Things like "A6", "D7" or "PB5" are more common. Putting strings in the example also helps people immediately see how thinks like "LED2" or named multi-pin connectors (such as Olimex UEXT) would fit into the system.

zolkis, 2016-11-11:
Right, for GPIO pin names both numbers and strings are accepted, as long as they make sense to the board/OS, so I will need to give example for string names as well. 
For GPIOPort names only strings are accepted.

zolkis, 2016-11-11:
> I would have more comments on proposed API, just need to read it in full and think out my replies.

That would be great help. Thank you!

pfalcon, 2016-11-14:
> This is definitely true. However, when reviewing the APIs, keep in mind the iot-js-api APIs may be implemented on more platforms than Zephyr. So I am trying to come up with an API that enables the Zephyr use cases while allowing others as well.

I understand that, and don't envy you ;-) - we had similar API design issue in another project, and it's *very* hard to make everyone interested satisfied. So, I'd be approaching it from practical side this time, just looking for things which make it harder for Z.js comparing to the current API. (Still in the backlog due to Zephyr itself going into 1.6RC, sorry).

Anyway @zolkis , I assume you don't have concerns with notation of `"GPIO_1.21"` as proposed above and it fits your new API well.

But then @poussa , would you have any issues? And while situation with implementation of new API is not exactly clear, I'd like to be on practical side and consider how to implement it in the current API, even as a proof of concept. In that regard, I updated the ticket description.

pfalcon, 2016-11-14:
@poussa : Actually, discussing changes theoretically is one thing, and being eyeball is another, so I implemented it: https://github.com/01org/zephyr.js/pull/438

poussa, 2016-11-14:
Thanks. I'll take a look tomorrow.

poussa, 2016-11-15:
I like the idea of using 2 dimensional string in to identify the device and pin. The real value is that Zephyr OS does the heavy lifting and ZJS does need to catch up and add support for each and every board. With that, I am suggesting we change the API spec to add this feature. 

We need to discuss if we can and/or want to keep the original API design idea of using the pin labels printed on the board or not. 

I will talk with @zolkis about that but I would encourage @pfalcon to continue with the implementation. I'll check the PR#438 in more detail tomorrow.

pfalcon, 2016-11-16:
Thanks! I'm on vacation thru next week (and that's why I wanted to post some patch so people could look at something in the meantime), will return to this after that.

grgustaf, 2018-01-08:
I believe with recent changes this is finally adequately addressed; closing.

closed 2018-01-08 by grgustaf