intel/zephyr.js #372: W3C Sensors API

baleboy, 2016-11-04:
Sensors API following W3C spec (https://www.w3.org/TR/2016/WD-generic-sensor-20160830/)

baleboy, 2016-11-04:
When we get to actually do this, it should probably be split in one feature per sensor.

kenchris, 2016-11-04:
This is happening... PRs already for accel/gyro

baleboy, 2016-11-04:
links?

haoxli, 2016-11-07:
https://github.com/01org/zephyr.js/pull/353

xiaolongx, 2017-01-13:
Basic feature is ready, verified with Samples: [Accelerometer.js](https://github.com/01org/zephyr.js/blob/zjs-0.2/samples/Accelerometer.js), [AmbientLight.js](https://github.com/01org/zephyr.js/blob/zjs-0.2/samples/AmbientLight.js),[Gyroscope.js](https://github.com/01org/zephyr.js/blob/zjs-0.2/samples/Gyroscope.js), but the issue still exist on Accelerometer.js #504 and AmbientLight.js #531 . 
Commit: cd4e16a
Hardware: Arduino_101

kenchris, 2017-01-16:
It seems that the Quark SE C1000 chip using in the Arduino 101 might have CPU core temperature exposed as a sensor as well:

http://click.intel.com/edc/evaluation-kits-current/intel-quark/intel-quark-se-microcontroller-evaluation-kit-c1000.html
http://www.intel.com/content/www/us/en/embedded/products/quark/mcu/se-soc/overview.html

```Temperature sensor (on Intel® Quark™ processor core)```

grgustaf, 2017-01-20:
Included in 0.2, @jimmy-huang can look at the CPU temp for 0.3.

qiaojingx, 2017-02-07:
Verified with samples: [Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js), [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js), [Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/Gyroscope.js), basic feature is ready. The known issues Accelerometer.js #504 and AmbientLight.js #531 won't be fixed in zjs-0.2 branch([ZJS 0.2 releases](https://github.com/01org/zephyr.js/releases)).
Commit: [eafdfe9](https://github.com/01org/zephyr.js/commit/eafdfe9e3f239e79e006fa7e2832eb4df5ba8305)
Hardware: Arduino 101

jimmy-huang, 2017-03-01:
closing this as this is implemented, I am keeping #96 open for Temperature sensor still

closed 2017-03-01 by jimmy-huang