intel/zephyr.js #373: Node.JS File API

baleboy, 2016-11-04:
Implement the Node.JS File API (Please add link to API spec here)

qiaojingx, 2017-03-23:
Basic feature of FsSync is ready on A101. FS is not supported on FRDM-K64F now. Verified result as below:

Samples | Platform | Known issues
---|---|---
[ FsSync.js ](https://github.com/01org/zephyr.js/blob/master/samples/FsSync.js) | Arduino 101 | Ready
[ FsAsync.js ](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js) | Arduino 101 | #829 
[ FsSync.js ](https://github.com/01org/zephyr.js/blob/master/samples/FsSync.js),[ FsAsync.js ](https://github.com/01org/zephyr.js/blob/master/samples/FsAsync.js) | FRDM-K64F | #803 

Commit ID a6c321a + patch #871

qiaojingx, 2017-03-31:
Basic features of FsSync and FsAsync are ready on A101, FS doesn't plan to support on FRDM-K64F at the monent(#803).
Commit ID 380261a
Platform: Arduino 101

jimmy-huang, 2017-04-24:
closing this as the basic feature is implemented.

closed 2017-04-24 by jimmy-huang