intel/zephyr.js #374: Node.JS HTTP API

baleboy, 2016-11-04:
Implement HTTP API as defined in Node.JS (please add link to API spec)

pfalcon, 2016-11-04:
@baleboy : I also will be interested to work on socket API support for Z.js, apparently starting with UDP as that's what support by native Zephyr IP stack so far. https://nodejs.org/api/dgram.html , http://www.hacksparrow.com/node-js-udp-server-and-client-example.html . Feel free to add a feature for that (I assume you're a PM for the project), or I'll submit an RFC later.

baleboy, 2016-11-04:
@pfalcon yes I'm doing PM duties on the project. It would be great to get feedback from the rest of the team before adding the feature, so why don't you go ahead with the RFC? Ideally you would put it in such a way that we can convert it to feature later, but we can also create a new issue.

poussa, 2016-11-04:
Z will have new IP stack and http lib soon. This feature should be built using those.

poussa, 2016-11-04:
Yes, we want more Node.JS APIs, UDP is low hanging fruit so +1 for that.

baleboy, 2016-11-04:
OK sounds like a no-brainer so I'll create the feature.