intel/zephyr.js #375: Node.JS Console API

baleboy, 2016-11-04:
Implement the node.JS Console API (link to API Spec TBD)

grgustaf, 2016-12-06:
The initial implementation landed in 0.1, improvements have already gone in for 0.2. @jprestwo can take a look and propose whether anything else should be added for 0.2 at this time.

jprestwo, 2016-12-14:
Looking [here](https://nodejs.org/api/console.html#console_console_time_label) it seems like we have everything (that is possible) except `console.time()` and `timeEnd()`. The rest of the API's that we don't have are not possible for us to implement at this time, e.g. `console.trace()`. But the time API's I could implement and get into 0.2.

haoxli, 2016-12-15:
Verify the implement status of Node.js Console API in commit bf8687586364d2dcb25d2e77da42769f302a11af
- [ ] ~~new Console()~~
- [ ] ~~console.assert~~
- [ ] ~~console.dir~~
- [x] console.error
- [x] console.info
- [x] console.log
- [ ] console.time
- [ ] console.timeEnd
- [ ] ~~console.trace~~
- [x] console.warn

jprestwo, 2016-12-15:
@haoxli, yes, and I have a PR for `console.time()` and `console.timeEnd()`. It would be easy to add `new Console()` because it will just return the existing console object, and `console.assert()` would be simple to add as well; I will go ahead and add those to my existing PR. So really, the only API's that we are not capable of doing are `console.dir()` and `console.trace()`.

xiaolongx, 2017-01-13:
`console.error`, `console.log` are ready, verified with Samples: [Console.js](https://github.com/01org/zephyr.js/blob/zjs-0.2/samples/tests/Console.js), and need to waiting #553 merge to verify `console.time()` and `console.timeEnd()`, but `new Console()`, `console.assert`, `console.info` and `console.warn` are not covered.
Commit: cd4e16a
Hardware: Arduino_101

grgustaf, 2017-01-20:
We'll expand somewhat for 0.3 but what's done is done for 0.2.

closed 2017-01-20 by grgustaf

haoxli, 2017-02-06:
@grgustaf Does it mean we need to create another feature id for Console API with 0.3 tag?