intel/zephyr.js #386: Node.JS UDP Socket API

baleboy, 2016-11-04:
https://nodejs.org/api/dgram.html

See also example: http://www.hacksparrow.com/node-js-udp-server-and-client-example.html

pfalcon, 2016-11-10:
As I mentioned elsewhere, I'd be interested to work on this (when more ground-level issues are tackled). So, if someone would look into this too, let's keep in touch.

poussa, 2016-11-10:
@pfalcon we don't have anybody working on this so it is yours.

pfalcon, 2016-11-10:
Thanks. With other background work (e.g. Zephyt 1.6 release and incompatibilities it brings) it may take some time before I may get to this, so leaving the above notice if anything changes.

jprestwo, 2016-11-10:
@pfalcon, have you started this yet? OCF just merged so I am open to making some headway here. @poussa did say he wanted HTTP as a priority but I think we are going to have to wait for the new IP stack (correct me if I'm wrong). Zephyr also has a simple echo client and server using UDP, so the current features for UDP do seem to work.

poussa, 2016-11-11:
@jprestwo Lets stick with the original plan. @pfalcon will take UDP (this) and you will do TCP+HTTP.

Both of you should be using the new IP stack which is in the 'net' branch of the Zephyr. The 'net' branch follows master closely and will be merged to master during this month. The current IP stack and APIs should not be used. Also, if you need any Zephyr services, please use the unified (V2) kernel APIs.

pfalcon, 2016-11-11:
@jprestwo : I didn't, I'm oscillating between Zephyr hardware drivers and being able to write desktop-style server with Z.js, and all things inbetween, so sorry that I'm making more RFCs than I can work on in a given period of time ;-). Generally, UDP support would be my pet project to keep elaborating my knowledge of Zephyr IP stack and gain some knowledge of Z.js' event/callback scheduling.

If you have nothing else in your queue and ready to work on TCP/HTTP (even though it's likely not fully ready upstream), then I may imagine you'd want to practice on UDP first, then feel free to take this. Otherwise, I agree with @poussa that with the feature list posted by @baleboy , there's doesn't seem to be lack of things to work on ;-).

jprestwo, 2016-11-11:
@poussa, sounds good, HTTP it is. I did look in the net branch and could not find any references to HTTP API's. We're we still going to go with trying to find an external library? Another option is using an already written JS module since @brianjjones is coming along with module support. Either way, we should discuss how we want to implement HTTP API's.

pfalcon, 2016-11-11:
Well, there definitely would be HTTP support in Zephyr: https://jira.zephyrproject.org/browse/ZEP-346 , https://gerrit.zephyrproject.org/r/#/c/4920/ .

Note that idea to be able to handle HTTP at a scripting language level is rather good (if not obvious): if you can program an HTTP server with your system, then it probably not too worse than multi-million datacenters which in large part run HTTP servers ;-). That's actually my reasoning for this ticket's task: I'd like to write HTTP server, for that I need TCP sockets, but TCP is not yet supported, so let's play with UDP for starters.

While the above reasoning is very attractive (namely, do it bottom up; and attractive for me), I can project what happens next: you find that scripting lingo impl is not the most efficient on memory usage (its fragmentation), and you want to optimize it with a C implementation.

poussa, 2016-11-11:
Yes, the http lib is there as @pfalcon pointed out.  It is Node's library. TCP will be, if not already, in the net branch soon. We should be able to start.

Being able to run REST API server with ZJS will be a killer feature!

pfalcon, 2017-01-22:
FYI, I finally started on this, and now even have something working. I travel next week, so it may take some time before it'll be posted though.

jprestwo, 2017-01-23:
@pfalcon, I don't know how far you've gotten but I have done some work in this area. Here is my branch: https://github.com/jprestwo/zephyr.js-1/tree/udp-zephyr

I am happy to continue, but I don't want to duplicate work (I probably should have said I started!). So far I have JS API's for opening a UDP socket, binding, and receiving data.

pfalcon, 2017-01-25:
@jprestwo : That's indeed a bit unfortunate, given the previous discussion on this ticket and comment from @poussa at https://github.com/01org/zephyr.js/issues/386#issuecomment-259883259 . I understand the concern - there were no progress for more than 2 months, and I was anxious about such situation too. But besides long HY holidays and me working on other things, there's a (well)known reason for slow progress: Zephyr's IP stack is, well, no up to par yet. So, whenever I touch anything networking in Zephyr, I get dragged into investigating or debugging some upstream issue.

The same happened again now, just touching Z network stack after a month of break, I hit 1, 2, 3 issues in row. Maybe I should have commented that I started on it earlier (I work on it for a week now), but I wasn't sure if I get to Z.js soon or will get dragged in the upstream stuff again (I try hard to "shift" upstream issues to other folks, yeah ;-) ).

Anyway, my plans would be: I have all basic sending/receiving working, just need to trace IPv6 path. But then there would be a lot of code refactoring. I already a week on it, so would need to finish it up - I can't say my boss "I spent a week on it, but changed my mind". I also have a talk at the upcoming OpenIoT comparing Z.js to other languages, and for that, I need more experience with Z.js, and well, that's me gaining that experience.

poussa, 2017-01-25:
Seems that you are both pretty much on the same state. We also talked about this yesterday and agreed that we wait for your (@pfalcon) response. My guidance would be that @jprestwo will focus now on TCP and HTTP and @pfalcon will get the UDP done.

pfalcon, 2017-01-26:
Thanks @poussa . I posted my WIP version as https://github.com/01org/zephyr.js/pull/659. While there're still things to do on my side, it generally works, and could use some feedback on macros and other patterns used.

baleboy, 2017-03-23:
According to @poussa this is implemented but only works on FRDM board at the moment.

baleboy, 2017-03-30:
As discussed with @poussa, we should make this work on A101 as well for the 0.3 release unless this is a huge amount of work.

haoxli, 2017-03-31:
I think it will be clearer if add label or comment for which targets plan to be supported for the feature.

qiaojingx, 2017-04-11:
Basic feature of UDP is ready on both A101 and FRDM-K64F. IPv4 is not expected to work on Arduino 101 since its using 6lowpan.
Commit ID cc2311a
Platform: Arduino 101/ FRDM-K64F

jimmy-huang, 2017-04-24:
closing this as the basic feature is implemented.

closed 2017-04-24 by jimmy-huang