intel/zephyr.js #397: [I2C.js] One more "WORLD" displays on LCD after running 3-4 hours

xiaolongx, 2016-11-09:
After running [I2C.js](https://github.com/01org/zephyr.js/blob/zjs-0.1/samples/I2C.js) 3-4 hours with commit [b1be3ba](https://github.com/01org/zephyr.js/commit/b1be3ba69cd5c53bdc69500d97529be9306af054) in zjs-0.1 branch, one more "HELLO" or "WORLD" displays on LCD screen, please refer to the attached images. 
Expect output: one "HELLO!" displays with orange in background, and one "WORLD!" displays with blue in background.

[image]
[image]

grgustaf, 2016-11-09:
Heh I just saw this same bug yesterday (except mine said HELLO!) and showed Brian. Basically what must be happening is that one out of thousands of I2C commands is getting dropped. It's hard to know whether that is in the Zephyr driver or hardware or just unavoidable. Seems doubtful that it could be in our code though.

grgustaf, 2016-11-09:
(That is, the I2C command to move the cursor back to the beginning of the line is the one that gets dropped.)

grgustaf, 2016-11-09:
I'll let Brian look into it. Not a high priority. :) But one thing we could do is try to estimate how often it happens 1/1000? 1/10k? Etc. By measuring the average time for the problem to occur. If we spit out a timestamp instead of the Hello and World then it might leave the evidence right on the LCD for us.

And then we'd want to approach Zephyr folks with the issue, probably by reproducing it in a native Zephyr app first.

grgustaf, 2016-11-09:
@xiaolongx, had you ever tried this before? Two nights ago was the first time I'd ever left I2C.js running overnight, so I assumed the problem might have always been there.

xiaolongx, 2016-11-10:
Yes, I tried in master branch(commit 2ff271b) for 6 hours, but cannot reproduce this issue, I will rerun again over night.

brianjjones, 2016-11-22:
I think this is a case of, I need to check the return when I write to I2C.  If it fails try again rather than going forward.  Basically it fails to move the cursor back, and then writes the text anyway.

brianjjones, 2016-11-30:
The fix for this has now been merged into master
https://github.com/01org/zephyr.js/pull/468

grgustaf, 2016-12-02:
That's not a fix, that's masking over the problem. Guess that slipped in, I didn't intend to merge that. :)

grgustaf, 2016-12-02:
I guess the difference is a bit subtle. In a product, sure you might do this to work around a bug. But we're enabling, we should be interested in monitoring and investigating the root cause of such a failure. So I would revert that part of the patch.

grgustaf, 2016-12-06:
It's not clear whether the bug is in hardware or Zephyr, or a timing / race condition type thing in ZJS, but that's less likely.

brianjjones, 2016-12-07:
It happens because a i2c_write call to zephyr fails.  Specifically the "shift cursor back to the beginning" write call.  The next word is then written at the end of the first, causing the double word.  The real fix would be to either a.) Have the I2C API be smart enough to try again if the write fails b.) Add a return value to the I2C API so that the javascript is aware it needs to try again.

brianjjones, 2016-12-14:
This should be fixed by https://github.com/01org/zephyr.js/pull/525 which will offer a return value on writes.  This way the javascript can request a write again should it fail.

grgustaf, 2016-12-14:
So you've actually been able to log that you do get a write failure reply from that function, once in a blue moon? It probably does make sense to have retry logic right in the API then. At least for just a single retry to cover flukes.

xiaolongx, 2017-01-12:
This issue reproduce on Arduino 101 branch with commit cd4e16a(zjs-0.2 branch).

qiaojingx, 2017-02-09:
This issue can not be reproduced after running over night on Arduino 101 branch with commit eafdfe9(zjs-0.2 branch).

xiaolongx, 2017-02-10:
Closed since it is not reproducible.

closed 2017-02-10 by xiaolongx