intel/zephyr.js #406: convert.sh requires uglifyjs

poussa, 2016-11-10:
If you don't have uglifyjs installed the new convert.sh fails to do proper job. 

Installing uglifyjs fixed the issues for me which I reported at https://github.com/01org/zephyr.js/pull/327

jprestwo, 2016-11-10:
This should fix the problem: https://github.com/01org/zephyr.js/pull/411

The issue was that newlines were not handled (because uglify did not output newlines).

jprestwo, 2016-11-11:
@poussa, I have confirmed that #411 fixes the problem, but could you give it a whirl as well, preferably on a Mac to make sure I didn't miss anything (with and without uglifyjs installed). I was also thinking, and it may be beneficial to allow a flag to be passed in which does not use uglifyjs even if it is installed, so the user can just run non-minified JS if so desired. What do you think about that?

jprestwo, 2016-11-14:
This has been merged and appears to fix the problem. We can close this issue.

closed 2016-11-14 by poussa