intel/zephyr.js #410: Zephyr 1.6.0-rc final upgrade issues

pfalcon, 2016-11-10:
Zephyr 1.6.0 is about to enter RC phase, and one of the big changes expected was "unified kernel", and it actually landed this week in master. This change was supposed to be mostly backward-compatible, but by various reasons it's actually not. I now retesting various components I'm working with against unified kernel, and gave Z.js a try too, and opening this ticket to notify the team about issues expected.

This ticket can probably be used for general 1.6.0 final upgrade discussion.

pfalcon, 2016-11-10:
~~~
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/main.c: In function 'main':
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/main.c:156:9: warning: '_legacy_sleep' is deprecated [-Wdeprecated-declarations]
~~~

This is an easy one to upgrade (but note that usage of deprecated API is error in Z.js CFLAGS, I had to patch that).

pfalcon, 2016-11-10:
~~~
In file included from /mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/zephyr/include/kernel.h:2388:0,
                 from /mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/zephyr/include/zephyr.h:20,
                 from /mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_pool.c:4:
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/zephyr/include/legacy.h:1788:1: note: expected 'struct k_mem_pool *' but argument is of type 'uint32_t {aka long unsigned int}'
 task_mem_pool_alloc(struct k_block *blockptr, kmemory_pool_t pool_id,
 ^
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/src/zjs_pool.c:208:22: error: 'struct k_mem_block' has no member named 'pointer_to_data'
     add_pointer(block.pointer_to_data, block.req_size);
                      ^
~~~

This looks much harder, and where I stop, submitting this report. Note that the above may be the case of Z.js trying to use private Z API.

pfalcon, 2016-11-10:
Also, to clarify, I wanted not just test unified kernel, but also try Z.js against a board added recently to Z master. In particular, to serve as a practical argumentation for https://github.com/01org/zephyr.js/issues/368. So, I'm currently blocked to do that (which is not big deal, I'm just taking a chance to say to that people (e.g. me) interested to add more board support for Z.js; actually, there's interested to just run Z.js on any board supported for Z, without a need to go thru extra hops).

grgustaf, 2016-11-15:
Okay, the memblock pointer thing - yeah when we were using that I pointed out to the Zephyr folks how much useless overhead there was in the k_block struct, and they must have done something about it. Looking back, we were a bit nuts to go optimize that on our end, we should have instead followed the progress in Zephyr itself - I need to learn to stop making that mistake. @jprestwo implemented that so he can fix it and hopefully we can just use the normal struct now.

We do need to get on this ASAP.

BlankRain, 2016-12-04:
I have the same problem..
``
glove@ubuntu:~/zephyr.js$ make
Found dependency: jerryscript
Found dependency: zephyr
Found dependency: ihex
cp prj.conf.base prj.conf
Creating C string from JS application...
(node) util.print is deprecated. Use console.log instead.
100% 0x3b ............................
make[1]: Entering directory '/home/glove/zephyr.js'
Using /home/glove/zephyr.js/deps/zephyr/boards/x86/arduino_101/arduino_101_defconfig as base
Merging /home/glove/zephyr.js/deps/zephyr/kernel/configs/unified.config
Merging prj.conf
make[2]: Entering directory '/home/glove/zephyr.js/deps/zephyr'
make[3]: Entering directory '/home/glove/zephyr.js/outdir/arduino_101'
  GEN     ./Makefile
 configuration written to .config
make[3]: Leaving directory '/home/glove/zephyr.js/outdir/arduino_101'
make[2]: Leaving directory '/home/glove/zephyr.js/deps/zephyr'
make[2]: Entering directory '/home/glove/zephyr.js/deps/zephyr'
make[3]: Entering directory '/home/glove/zephyr.js/outdir/arduino_101'
  GEN     ./Makefile
scripts/kconfig/conf --silentoldconfig Kconfig
  Using /home/glove/zephyr.js/deps/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
  CHK     misc/generated/sysgen/prj.mdef
  CC      src/main.o
/home/glove/zephyr.js/src/main.c: In function 'main':
/home/glove/zephyr.js/src/main.c:136:9: error: '_legacy_sleep' is deprecated [-Werror=deprecated-declarations]
         zjs_sleep(1);
         ^
In file included from /home/glove/zephyr.js/deps/zephyr/include/kernel.h:2906:0,
                 from /home/glove/zephyr.js/deps/zephyr/include/zephyr.h:20,
                 from /home/glove/zephyr.js/src/main.c:4:
/home/glove/zephyr.js/deps/zephyr/include/legacy.h:421:26: note: declared here
 extern void __deprecated _legacy_sleep(int32_t ticks);
                          ^
cc1: all warnings being treated as errors
/home/glove/zephyr.js/deps/zephyr/scripts/Makefile.build:183: recipe for target 'src/main.o' failed
make[4]: *** [src/main.o] Error 1
/home/glove/zephyr.js/deps/zephyr/Makefile:958: recipe for target 'zephyr-app-dir' failed
make[3]: *** [zephyr-app-dir] Error 2
make[3]: Leaving directory '/home/glove/zephyr.js/outdir/arduino_101'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/glove/zephyr.js/deps/zephyr'
/home/glove/zephyr.js/deps/zephyr/Makefile.inc:82: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/glove/zephyr.js'
Makefile:23: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2

```

pfalcon, 2016-12-04:
Zephyr.js was upgraded to 1.6.0-rc-something in https://github.com/01org/zephyr.js/pull/450 , closing.

closed 2016-12-04 by pfalcon