intel/zephyr.js #418: Fail to clone 'iotivity-constrained' directly via command: git update

xiaolongx, 2016-11-11:
Fail to clone 'iotivity-constrained' directly via commands in master branch (commit 1b9d779):
```
$ git clone https://github.com/01org/zephyr.js.git
$ cd zephyr.js
$ source zjs-env.sh 256
$ make update
......
Cloning into 'deps/iotivity-constrained'...
fatal: unable to access 'https://gerrit.iotivity.org/gerrit/p/iotivity-constrained.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: clone of 'https://gerrit.iotivity.org/gerrit/p/iotivity-constrained.git' into submodule path 'deps/iotivity-constrained' failed
Makefile:81: recipe for target 'update' failed
make: *** [update] Error 128
```
But, after setting .ssh/config following https://wiki.iotivity.org/how_to_use_gerrit step 1 & 2, I can clone iotivity-constrained from https://gerrit.iotivity.org/gerrit/p/iotivity-constrained.git, could you please provide the regular commands?

pfalcon, 2016-11-11:
I just made a fresh pull and make update, everything worked for me:

~~~
$ git pull --rebase upstream master
remote: Counting objects: 109, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 109 (delta 65), reused 64 (delta 64), pack-reused 39
Receiving objects: 100% (109/109), 40.11 KiB | 0 bytes/s, done.
Resolving deltas: 100% (67/67), completed with 41 local objects.
From https://github.com/01org/zephyr.js
 * branch            master     -> FETCH_HEAD
   ae9fc8c..1b9d779  master     -> upstream/master
First, rewinding head to replay your work on top of it...
Fast-forwarded master to 1b9d779d2fb298ea8c50e4a491706c5f152d691c.
pfalcon@x230:~/projects-3rdparty/Embedded/Zephyr/zephyr.js$ make update
Submodule 'deps/iotivity-constrained' (https://gerrit.iotivity.org/gerrit/p/iotivity-constrained.git) registered for path 'deps/iotivity-constrained'
Cloning into '/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/iotivity-constrained'...
Submodule path 'deps/iotivity-constrained': checked out '1b7513395ddf96610bf02d40054a28ac98ffa523'
Submodule path 'deps/zephyr': checked out '9cbd41ab13bf2b48bcfb6e3e97b4d59020004707'
Submodule 'deps/tinycbor' (https://github.com/01org/tinycbor.git) registered for path 'deps/tinycbor'
Cloning into '/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/iotivity-constrained/deps/tinycbor'...
Submodule path 'deps/tinycbor': checked out '0a78b71322fe2f54b342c1b6ea66ce31da41383e'
~~~

pfalcon, 2016-11-11:
~~~
fatal: unable to access 'https://gerrit.iotivity.org/gerrit/p/iotivity-constrained.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
~~~

Was this reproducible when run multiple times? I'd say this looks like a usual non-deterministic one-time network glitch.

jprestwo, 2016-11-11:
@pfalcon, @xiaolongx, I am not seeing this either, so I think your correct, it was probably a one-time network problem. @xiaolongx, are you still seeing this issue? If not we can close this issue.

grgustaf, 2016-11-15:
Yes, I agree. Please close.

xiaolongx, 2016-11-15:
Able to clone 'iotivity-constrained', maybe it is network issue sometimes, close

closed 2016-11-15 by xiaolongx