intel/zephyr.js #424: Building linux with VARIANT=debug must be done after a clean

jprestwo, 2016-11-11:
If you build in release mode (default), then afterwards try and do a debug build `VARIANT=debug`, the source files are not recompiled with the debug flags. In order to build in debug mode you have to `make clean`, then rebuild with `VARIANT=debug` again. This should not be the case. The Makefile should know that the build flags have changed, and rebuild the source files using the debug/release flags depending on what is passed in and rebuild source files if necessary.

grgustaf, 2016-11-11:
As I commented in IRC, instead build debug and release to separate places like outdir/linux/debug and outdir/linux/release.

Also maybe we should put jslinux there too instead of in the root dir; I'll leave that up to you.

This same technique probably avoids the need for last-build file now?

jprestwo, 2016-11-14:
@grgustaf, good idea. I hope that it will avoid the .last_build file. The original reason I added that was so that it would clean between Zephyr and Linux builds... but in truth, there is really no reason for that because they are using different build systems/compilers so there should never be a "stale" file problem.

jprestwo, 2016-11-14:
@grgustaf, I have implemented your suggestion of putting everything under outdir/linux. That change can be seen here: https://github.com/01org/zephyr.js/pull/436

This makes the build tree much cleaner as it doesn't leave a bunch of object files throughout the build tree because it copies everything to outdir/linux/<variant>, similar to how Zephyr works. The final executable is located under outdir/linux/<variant>. I moved it in order to make it possible to do a release build and debug build while being able to keep both binaries. Not sure of the use case, but its nice if you want to run your previous debug build without having to completely rebuild.

pfalcon, 2016-11-14:
> The final executable is located under outdir/linux/. I moved it in order to make it possible to do a release build and debug build while being able to keep both binaries.

I'd say that makes things less convenient. But then anyone can have their own symlinks in their working copies. Except for newcomers who may have no idea where to look for stuff, unless described in the README.

grgustaf, 2016-11-15:
Yeah, that was the bit I was iffy about too. I'm okay with it if you want to leave jslinux in the root dir. I don't like it to be too cluttered but I removed a bunch of build artifact stuff recently anyway.

grgustaf, 2016-12-06:
So @jprestwo make sure some basic instructions and the jslinux location are mentioned in the README at least I guess? Is it fixed?

jprestwo, 2016-12-06:
@grgustaf, yes this is fixed now, building in either debug or release mode will cause an automatic clean. But I do still need to add the documentation to the README.

grgustaf, 2017-01-18:
@jprestwo, checking on that documentation?

jprestwo, 2017-01-20:
@grgustaf, documentation is in the top level README about the VARIANT flag.

grgustaf, 2017-01-20:
Included in 0.2, closing.

closed 2017-01-20 by grgustaf