intel/zephyr.js #432: console.log causes zephyr to fail

jbeard4, 2016-11-14:
I have been able to deploy the traffic light example to my arduino 101 board, and this is working well. I am trying to get serial logging to work with the USB to TTL Serial Cable for the Raspberry Pi: https://www.adafruit.com/products/954?gclid=CJPypeGcptACFU1LDQodoTADWA

So far, I have not been able to see any serial output from this cable when connected to the arduino 101. I have tried modifying the traffic light by adding statement `console.log(elapsed)` inside of setInterval. However, I still do not see any serial output, and this appears to cause the program to stop immediately, when the red LED is illuminated. When I add `print(elapsed)` instead of `console.log(elapsed)`, the program works normally, and cycles through the three LEDs, but I still do not see any serial output. The TX and RX LEDs also do not illuminate.

Why does console.log seem to break the traffic light example? And what is the best way to troubleshoot the serial output?

Thank you.

jprestwo, 2016-11-14:
The only difference between `print` and `console.log` is that `print` uses `printf()` and `console.log` uses `fprintf()` (so it can selectively print between stderr and stdout). Apart from `console.log` doing argument parsing, that is the only difference so they should behave the same.

Regarding serial output, the Arduino 101 is a bit different than other boards (this applies to the 'uart' module). If your using `uart.write()`, it is actually going over USB using ACM. This port can be opened the same as you do with a USB-to-serial adapter but the name of the port is generally `/dev/ttyACM0`. In case you didn't know, `console.log` and `print` both go to `/dev/ttyUSB0`. With the UART module, you do not need a USB-to-Serial adapter, just plug the board in through its USB port. Note that `/dev/ttyACM0` will only show up after the board has started up and initialized.

grgustaf, 2016-11-15:
@jbeard4, thanks for trying our project out! If it's possible for you to join us in #zjs on Freenode IRC someone would be happy to help you debug this.

First, the hardware - we've only been using this one particular TTL-232R-3V3 USB to TTL serial cable, so other cables might vary. In our case, we wire the orange pin to IO0 (RX), the green pin to IO1 (TX), and the white pin to GND (just past IO13). So I think you want to wire your green TX into IO0 (RX), your white RX into IO1 (TX), and your black into GND (just past IO13), leaving your power disconnected.

Then here's how I debug it. First, unplug the serial cable, then see what relevant ttys I have showing up:
```bash
$ ls /dev/tty[AU]*
```

Then plug it back in and run the command again to see what is new:
```bash
$ ls /dev/tty[AU]*
/dev/ttyUSB2
```

In my case, ttyUSB2 showed up.

Then I am able to run this screen command:
```bash
$ screen /dev/ttyUSB2 115200
```

... and it will clear the screen and wait for input. If it doesn't, it may not detect anything there. Try unplugging the USB on the host side and plugging it back in?

Then, since you already had the TrafficLight.js sample working you should be able to just use the normal HelloWorld sample to get some console output. You can build the X86 image and then try to flash it like this:

```bash
$ make all
$ make dfu-all
```

Reset the device a few seconds before running `make dfu` as you know already. As long as the progress bar gets to 100% it should have completed, otherwise try again. Then reset the device again with MASTER RESET button and in about 6s it should come up with this output:

```
ipm_console0: 'Sensor core running ZJS ARC support image'

Hello, ZJS world! 
```

If not, you can always try swapping your RX and TX pins to make sure they weren't backwards. :)

If it's still not working, then I would go back to the Zephyr hello world sample and see if it works. For example:

```bash
$ cd deps/zephyr/samples/hello_world/nanokernel
$ make BOARD=arduino_101
$ dfu-util -a x86_app -D outdir/arduino_101/zephyr.bin 
```

If that works, you will see:
```
Hello World! x86
```

Then really ours should just work so maybe retrace the steps. But if not, maybe try consulting [Zephyr docs online](https://wiki.zephyrproject.org/view/Arduino_101). However, if you give us more detail or come talk to us on IRC we'll try to help figure it out.

Please let us know if you figure out anything missing in our README that could have helped solve your problem. Thanks!

jbeard4, 2016-11-15:
Thanks for the quick replies. I have ordered the TTL-232R-3V3 USB to TTL serial cable, and I will follow up on this issue once I have tested the cable.

jbeard4, 2016-11-18:
I got the TTL-232R-3V3 USB to TTL serial cable, and logging is now working well.

closed 2016-11-18 by jbeard4