intel/zephyr.js #434: [REG][I2C.js] It only display "HELLO" and "WORLD" without background color in GroveLCD screen

xiaolongx, 2016-11-14:
In commit 916cb71, flashing the [I2C.js](https://github.com/01org/zephyr.js/blob/master/samples/I2C.js) with below commands, no background color in GroveLCD screen, only "HELLO" and "WORLD" displays.
```
$ make all JS=samples/I2C.js
$ make dfu-all
```
If flashing [GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js) firstly, and then flash [I2C.js](https://github.com/01org/zephyr.js/blob/master/samples/I2C.js), it only display background without "HELLO" and "WORLD" in GroveLCD screen.

Sample | Commit | Result
---|---|---
I2C.js | 2ff271b | PASS 
I2C.js | 916cb71 | FAIL

brianjjones, 2016-11-22:
It turns out there were some registers that needed to be set for it to turn on.  And it turns out the Grove_LCD module has been doing this all along.  So when @jimmy-huang changed the arc build to only include the specified modules, the I2C demo broke because it didn't include the Grove_LCD module.  Basically it was always broken I just didn't know it :)  I'll fix it and push a PR.

brianjjones, 2016-11-30:
The fix for this is now in master

grgustaf, 2016-12-06:
@xiaolongx please verify

xiaolongx, 2016-12-06:
This issue cannot reproduce with commit a162db1 on both Arduino_101 and FRDM_K64F, close

closed 2016-12-06 by xiaolongx