intel/zephyr.js #444: ROM of sample increasing obviously on Arduino 101 & FRDM_K64F in master-916cb71

hanyia, 2016-11-15:
In WW47 PnP test against master-916cb71, found ROM increasing obviously on Arduino 101 & FRDM_K64F. Detailed result as attachment.

hanyia, 2016-11-15:
Or data from Matrix: 
http://metrix.sh.intel.com?bvb_project_id=2&device_id=68&dist_id=21&sessionA=4594&sessionB=4569&session_type_id=1&flag_toggle=1&page=data

hanyia, 2016-11-15:
[Weekly PnP Testing Report for ZJS on Arduino 101  FRDM_K64F (master-916cb71) - WW47.zip](https://github.com/01org/zephyr.js/files/591020/Weekly.PnP.Testing.Report.for.ZJS.on.Arduino.101.FRDM_K64F.master-916cb71.-.WW47.zip)

jimmy-huang, 2016-11-15:
I've backed tracked all the commits, it seems the two offending commits that adds significant ROM size are afe5b7 and ae9fc8, which adds like 16k of space, @jprestwo can you take a look at them to see if these PRINTs should only be enabled for debug builds, not release?

jprestwo, 2016-11-15:
@jimmy-huang, regarding the error prints, we were talking on IRC and it was decided that error prints should be turned on always so that if an error happens in a release build it gets seen. I am torn between leaving them out or keeping them in for release builds so we need to discuss this. Regarding the console.log change, that does add a flag which (I think) enables floating point in SW which may increase the ROM size. If this significantly increases the ROM footprint then we will just have to go back to how it was before and print the integer version of the number e.g.
```
[Float ~42]
```
Instead of
```
42.345423
```

poussa, 2016-11-15:
Can we put the proper float support behind a built time option. It is good to have both options; proper behavior and minimal ROM usage depending which one you value more.

jimmy-huang, 2016-11-15:
yeah, we also talked about in release mode, instead of printing strings in errors, print some kind of error code that user can then references back to the error instead, which further reduce the rom size.

jprestwo, 2016-11-16:
@jimmy-huang, yes, but I am kinda thinking that this problem has to do with the floating point support. The error strings do not seem to add up to ~16k of space. And even then, we would still likely print something like:
```
Error: <code>
```
So we still have to store the "Error:" string which will take up some space. In all reality, a release build *should* even compile out serial communication. A release build would go into a final product, which would not have a serial console/output, so my vote is to turn OFF error prints in a release build. The last project I worked on used "debug levels" in order to compile in/out certain statements. You would compile with DEBUG_LEVEL={0-5}, where 5 is all debug prints and 0 is a release builds. Right now we would only have 0-2 because we just have ZJS_PRINT, DBG_PRINT, and ERR_PRINT. We also should be using JavaScript error objects instead of simple prints, so the error is propagated up to JS, that way the developer sees the prints in their own if blocks catching errors.

grgustaf, 2016-12-06:
PR #466 should have largely addressed this. @hanyia, please verify.

hanyia, 2016-12-07:
From latest test report, ROM of sample reduce 10K+, so I think we could close this issue.

grgustaf, 2017-01-20:
Sounds good, thanks.

closed 2017-01-20 by grgustaf