intel/zephyr.js #446: [REG] Fail to receive 4 characters in UART.js with commit 916cb71

xiaolongx, 2016-11-15:
In commit 916cb71, input 3 characters, the data can be received successfully in [UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js). 
But, input 4 characters and click "Enter", the data cannot be received.

Log Info:
```
UART write succeeded, echoing input.
AT
AT
ATta recv: AT
A
AT
Data recv: AT
~
~~
a
Data recv: a
a
aa
aaa
Data recv: aaa
a
aa
aaa
aaaa
```
This issue cannot reproduce on last weekly testing:

Sample | Commit | Result
---|---|---
UART.js | 2ff271b | PASS 
UART.js | 916cb71 | FAIL

jprestwo, 2016-11-16:
@xiaolongx, first off, your machine is treating the device as a modem, that is why you are seeing "AT" and "~~" characters. This can be turned off by uninstalling a package, which I cannot remember. @sergioamr, was the one who instructed me how to fix this. BUT, I still suspect this is not the problem if it simply just stops receiving characters.

@grgustaf, @jimmy-huang, you can assign this one to me.

xiaolongx, 2016-12-06:
This issue cannot reproduce with commit a162db1 on both Arduino_101 and FRDM_K64F, close

closed 2016-12-06 by xiaolongx