intel/zephyr.js #447: [Timers] Instruction Access Violation in FRDM_K64F

xiaolongx, 2016-11-15:
After #310 fix, flashing [test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js) on commit 916cb71, and then rebooting frdm_k64f, instruction access violation occurs.

Error logs:

```
PASS - clearInterval: intervalID does not exist
PASS - clearTimeout: timeoutID does not exist
PASS - setInterval: set delay and optional arg
***** MPU FAULT *****
  Executing thread ID (thread): 0x20006fd8
  Faulting instruction address:  0xbdffdef6
  Instruction Access Violation

Fatal fault in task ! Aborting task

```
This issue cannot reproduce on Arduino 101

Platform | Tests | Commit | Result
---|---|---|---
Arduino 101 | test-timers.js | 916cb71 | PASS 
FRDM_K64F |test-timers.js | 916cb71 | FAIL

jprestwo, 2016-11-16:
@xiaolongx, this appears to be a new-ish problem. I was never able to get the FRDM board to connect over OCF but never saw this error. My issue, I think, was because I was trying to run the sample over a corporate network, so who knows how the router is configured.

I will look at this problem and see if I can get it back to how it was originally.

xiaolongx, 2016-11-28:
New error logs on FRDM_K64F with commit 5844f9f, it is caused by commit 7d861b2
```
PASS - clearInterval: intervalID does not exist
PASS - clearTimeout: timeoutID does not exist
PASS - setInterval: set delay and optional arg
***** Reserved Exception ( -16) *****
Fatal fault in task ! Aborting task.
```

xiaolongx, 2016-12-20:
New error logs on FRDM_K64F with commit 5a25755
```
PASS - clearInterval: intervalID does not exist
PASS - clearTimeout: timeoutID does not exist
PASS - setTimeout: set 20 arg for timeout
PASS - setTimeout: set delay and optional arg
PASS - setTimeout: set delay
PASS - setInterval: set delay and optional arg
***** Reserved Exception ( -16) *****
Fatal fault in essential thread! Spinning...
```

jprestwo, 2017-01-04:
An update here:

I tried doing a git bisect to see where this issue came up, but going back several months it looks like this issue was always present on the FRDM board. Depending on where in the history you are the actual fault will change types (bus, mpu, and currently "Reserved Exception") but in every case it faults. I still have not gotten to the bottom of it but I just wanted to update everyone and let you know this is not a regression, unless someone has seen it work in the past.

xiaolongx, 2017-01-06:
This issue cannot reproduce on FRDM-K64F with commit 2345f96, close

closed 2017-01-06 by xiaolongx