intel/zephyr.js #449: [Makefile] arc/src/Makefile missing when run 'make all' first time

haoxli, 2016-11-15:
Error always occurs when run `make JS=/path/to/filename.js all` first time. It reports arc/src/Makefile is missing.
```
$ git clone https://github.com/01org/zephyr.js
$ cd zephyr.js
$ source zjs-env.sh
$ make update
$ source deps/zephyr/zephyr-env.sh
$ make JS=samples/TrafficLight.js all
```
Error log:
```
/work/zephyr.js/deps/zephyr/scripts/Makefile.build:45: /work/zephyr.js/arc/src/Makefile: No such file or directory
make[4]: *** No rule to make target '/work/zephyr.js/arc/src/Makefile'.  Stop.
/work/zephyr.js/deps/zephyr/Makefile:946: recipe for target 'zephyr-app-dir' failed
make[3]: *** [zephyr-app-dir] Error 2
make[3]: Leaving directory '/work/zephyr.js/arc/outdir/arduino_101_sss'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/work/zephyr.js/deps/zephyr'
/work/zephyr.js/deps/zephyr/Makefile.inc:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/work/zephyr.js/arc'
Makefile:191: recipe for target 'arc' failed
make: *** [arc] Error 2
```
After run `make JS=/path/to/filename.js`, 'make all' won't report the error.

jprestwo, 2016-11-16:
@haoxli, sounds like a corner case build issue. @grgustaf, @jimmy-huang, either of you can assign this one to me and ill take a look.

xiaolongx, 2016-12-27:
This issue cannot be reproduced with commit e0d4559.

closed 2016-12-28 by haoxli