intel/zephyr.js #451: [Accelerometer.js][Gyroscope.js] Fail to get the x,y,z value and catch errors sometimes

xiaolongx, 2016-11-16:
Running [Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js) and [Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/Gyroscope.js) after 1-2 seconds, catch errors in serial log. 

Error logs:
```
x=0.53037 y=0.953175 z=0.07454999999999999
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww47-1/zephyr.js/src/zjs_callbacks.c:341 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
rotation (rad/s):  x=0.669885 y=0.410025 z=0.09[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww47-1/zephyr.js/src/zjs_callbacks.c:341 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
ephyr.js/src/zjs_rotation (rad/s):  x=0.531[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww47-1/zephyr.js/src/zjs_callbacks.c:341 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
xiaolong/ZJS/ww47-1/zephyr.js/src/zjs[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww47-1/zephyr.js/src/zjs_callbacks.c:341 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
rotati[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww47-1/zephyr.js/src/zjs_callbacks.c:341 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
```

Branch | Build ID | Date | Result
---|---|---|---
master | 870d928 | Nov 15, 2016 | FAIL

jimmy-huang, 2016-11-16:
yeah, this was a due to the ring buffer filling up too fast due to the event firing rapidly, so events gets dropped, it's a known issue, but it wasn't printing before now it's printing.   I have a new patch to use trigger mode to set the sampling frequency to a slower value to 50Hz which will eliminate this isse, re-test after #453 is merged.

xiaolongx, 2016-11-25:
Retest on arduino 101 with commit 2a9bbc7(#453 is merged), fail to get the x,y,z value
Error logs:
```
Accelerometer test...
ipm_console0: 'Sensor core running ZJS ARC support image'
state: activating
state: activated
```

```
Gyroscope test...
ipm_console0: 'Sensor core running ZJS ARC support image'
state: activating
state: activated
```

jimmy-huang, 2016-11-28:
#483 should fix x,y,z not updated issue.

xiaolongx, 2016-11-30:
QA cannot reproduce this issue after #483 is merged

grgustaf, 2016-12-06:
Closing based on QA verification.

closed 2016-12-06 by grgustaf