intel/zephyr.js #456: [GroveLCD] Keyword invalid with setInputState function

cuiyanx, 2016-11-17:
"GroveLCD.GLCD_IS_SHIFT_DECREMENT" and "GroveLCD.GLCD_IS_ENTRY_RIGHT" that was keyword with setInputState function in [grove_lcd.md](https://github.com/01org/zephyr.js/blob/master/docs/grove_lcd.md) can be set and get, but that is no effect at GroveLCD.

Target Device | Branch | Build ID | Date | Result
---- | ---- | ---- | ---- | ----
Arduino 101 | master | 916cb71 | Nov 17,2016 | FAIL
FRDM_K64F | zjs-0.1 | b9f4c5d | Nov 17,2016 | FAIL

grgustaf, 2016-12-06:
So at least figure out what these are supposed to do and document it if it's a known limitation.

jimmy-huang, 2017-03-27:
well, these flags are supposed to change the text orientation, so GLCD_IS_ENTRY_LEFT | GLCD_IS_SHIFT_DECREMENT would be for romance languages where text flows from left to right, and GLCD_IS_ENTRY_RIGHT | GLCD_IS_SHIFT_INCREMENT would set it sot he text flos from right to left, but yeah, they are not working as expected, and I tried the Zephyr sample, it's also not working, I think this is a upstream bug.

jimmy-huang, 2017-03-27:
bug filed upstream https://jira.zephyrproject.org/browse/ZEP-1971

cuiyanx, 2017-03-28:
Will we fix it or wait for this bug fixed by zephyr JS ?

jimmy-huang, 2017-03-29:
This is not a bug on our side, I think, so I am not sure how we can fix this.  At least wait to see if Zephyr can confirm that this is upstream bug.

jimmy-huang, 2017-04-03:
According to Zephyr's response on this bug, this is currently not implemented for the Grove LCD and there's no plan to add this, so I think we should drop support for this and update the API accordling.

commit e1824de3 (2017-04-10): [glcd] Drop unsupported Grove LCD input state API (#1001)

The Zephyr Grove LCD driver doesn't support text flow direction
set using LCD_IS_ENTRY_RIGHT, GLCD_IS_SHIFT_INCREMENT flags.  So
there's no need to expose the setInputState and getInputState
APIs.  Removing these until support is added in upstream.

Fixes #456
closed 2017-04-10 by grgustaf