intel/zephyr.js #463: [Ashell] Fail to enable the shell compilation with ‘make DEV=ashell’

xiaolongx, 2016-11-21:
Fail to enable the shell compilation with command: 
`make DEV=ashell`
Error Logs with commit 6037375:
```
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/bin/iamcu-poky-elfiamcu/../../libexec/iamcu-poky-elfiamcu/gcc/i586-poky-elfiamcu/5.2.1/real-ld: .tmp_zephyr.prebuilt section `initlevel' will not fit in region `ROM'
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/bin/iamcu-poky-elfiamcu/../../libexec/iamcu-poky-elfiamcu/gcc/i586-poky-elfiamcu/5.2.1/real-ld: region `ROM' overflowed by 528 bytes
collect2: error: ld returned 1 exit status
/home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/deps/zephyr/Makefile:840: recipe for target '.tmp_zephyr.prebuilt' failed
make[3]: *** [.tmp_zephyr.prebuilt] Error 1
make[3]: Leaving directory '/home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/outdir/arduino_101'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/deps/zephyr'
/home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/deps/zephyr/Makefile.inc:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js'
Makefile:40: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2
```

Branch | Build ID | Date | Result |
---|---|---|---
master | 6037375 | Nov 21,2016 | FAIL
master | f9791d9 | Nov 21,2016 | FAIL
master | d9d72cc | Nov 21,2016 | PASS

brianjjones, 2016-11-21:
Yep, this is a known issue.  It will be fixed by PR #462

grgustaf, 2016-12-06:
@xiaolongx @haoxli please retest

xiaolongx, 2016-12-06:
This issue cannot reproduce with commit a162db1 on both Arduino_101 and FRDM_K64F, close

closed 2016-12-06 by xiaolongx