intel/zephyr.js #465: [Samples] Callback does not exist error occurs after clearing interval

xiaolongx, 2016-11-22:
Two error messages occurs in [Timers.js](https://github.com/01org/zephyr.js/blob/master/samples/Timers.js) with commit 6037375, and also cannot output `Interval (no args) #1`. 

Serial Logs:
```
Starting Timers example...
Interval (args) #1 arg1: 1 arg2: 2
Interval (no args) #2
Interval (args) #2 arg1: 1 arg2: 2
Interval (no args) #3
Interval (args) #3 arg1: 1 arg2: 2
Interval (no args) #4
Interval (args) #4 arg1: 1 arg2: 2
Interval (no args) #5
Timeout, clearing interval
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/src/zjs_callbacks.c:435 zjs_call_callback(): callback does not exist: 1
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww48/master/zephyr.js/src/zjs_callbacks.c:435 zjs_call_callback(): callback does not exist: 0
```
But if we can increase 10ms delay in i2(setInterval) and setTimeout 
```
// Test with arguments
var i2 = setInterval(function(a, b) {
    console.log("Interval (args) #" + count + ' arg1: ' + a + ' arg2: ' + b);
    count++;
}, 1010, 1, 2);

setTimeout(function(a, b) {
    console.log("Timeout, clearing interval");
    clearInterval(a);
    clearInterval(b);
}, 5010, i1, i2);
```
We can got below output and no error messages:
```
Starting Timers example...
Interval (no args) #1
Interval (args) #1 arg1: 1 arg2: 2
Interval (no args) #2
Interval (args) #2 arg1: 1 arg2: 2
Interval (no args) #3
Interval (args) #3 arg1: 1 arg2: 2
Interval (no args) #4
Interval (args) #4 arg1: 1 arg2: 2
Interval (no args) #5
Timeout, clearing interval
```

jprestwo, 2016-12-15:
This is not actually an error (the callback service function just thinks there is a callback, which was cleared, and its checking for it and bailing out and printing that error). Its a funky timing thing, but I will look at it and see if maybe we are not cleaning up enough of "something" in callbacks. If its it turns out to be a common/expected code path I may just have to remove the print.

The main point I wanted to make is that nothing is actually going wrong as far as I can tell, we just need to change some logic around.

xiaolongx, 2016-12-20:
close based on comment above.

closed 2016-12-20 by xiaolongx