intel/zephyr.js #469: [OCF] "Imprecise data bus error" occurs when running OcfClientUpdate.js on FRDM-K64F

xiaolongx, 2016-11-23:
Flashing [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js), and then rebooting frdm_k64f, "Imprecise data bus error" occurs.

OcfClientUpdate.js on FRDM_K64F:
```
net: net_tx_fiber (0x200108ac): Starting TX fiber (stack 1024 bytes)
net: net_rx_fiber (0x200104ac): Starting RX fiber (stack 1024 bytes)
net: net_timer_fiber (0x20010cac): Starting net timer fiber (stack 1536 bytes)
net: net_driver_slip_open (0x2000efec): Initialized slip driver
oc_main: Stack successfully initialized
Started OCF client
***** BUS FAULT *****
  Executing thread ID (thread): 0x2000efec
  Faulting instruction address:  0x0001bc8c
  Imprecise data bus error
Fatal fault in task ! Aborting task.
```
OcfClientUpdate.js on jslinux:
```
oc_main: Stack successfully initialized
Started OCF client
Outgoing message to [ff02:0000:0000:0000:0000:0000:0000:0158]:5683
Sent 51 bytes
Incoming message from [fe80:0000:0000:0000:0478:bcca:8f91:07f8]:58439
```

Branch | Build ID | Date | Result |
---|---|---|---
master | 6037375 | Nov 23,2016 | FAIL

grgustaf, 2016-12-06:
I'm not sure OCF is really "done" and ready for testing yet, but I'll let @jprestwo comment.

jprestwo, 2016-12-06:
@grgustaf is correct, I am not even able to run the iotivity-constrained samples on the K64F board without running into errors. We should leave this open, but I think the problem has something to do with iotivity-constrained rather than how we are using it. Iotivity-constrained has also not been updated to the unified kernel, so I am waiting on that before trying to work around these issues.

jprestwo, 2017-04-04:
@xiaolongx, I don't see this behavior on master, could we check and see if this is relevant anymore.

closed 2017-04-07 by xiaolongx