intel/zephyr.js #477: [I2C] bus and speed are not a readonly properties

haoxli, 2016-11-25:
```
dictionary I2CInit {
    octet bus;
    I2CBusSpeed speed;
};

interface I2CBus {
    // has all the properties of I2CInit as read-only attributes
};
```

bus and speed  should be readonly properties, but actually it can be modified.

```
var i2cDevice = i2c.open({ bus: 0, speed: 100 });
var busValue = i2cDevice.bus;
var speedValue = i2cDevice.speed;
i2cDevice.bus = busValue + 1;
i2cDevice.speed = speedValue + 1;
console.log(i2cDevice.bus);
console.log(i2cDevice.speed);
```

The expected result are 0 and 100, but print 1 and 101.

Branch | Build ID | Date | Result
--- | ---|---|---
master | commit 2a9bbc7 | Nov 25, 2016 | Fail

grgustaf, 2016-12-23:
Fixed by #576. Improves pass rate of test-i2c.js to 12/12.

xiaolongx, 2017-01-06:
This issue cannot reproduce on FRDM-K64F with commit 2345f96

closed 2017-01-10 by haoxli