intel/zephyr.js #482: [REG] Event callback failed with commit 5844f9f

xiaolongx, 2016-11-28:
After flashing the samples with commit 5844f9f, three kinds of samples failed, and they were caused by commit 7d861b2:
1. GPIO: onchange does not work
[Button.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/digital/Button.js), [ButtonEnable.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/ButtonEnable.js), [AutoButton.js](https://github.com/01org/zephyr.js/blob/master/samples/AutoButton.js), [ButtonLEDs.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDs.js), [ButtonLEDsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDsAsync.js), [TwoButtons.js](https://github.com/01org/zephyr.js/blob/master/samples/TwoButtons.js), [Spaceship2.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/starterkit/Spaceship2.js)
2. UART: no serial log output via /dev/ttyACM0
[UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)
3. BLE: Fail to find Arduino 101 via BLE
[BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js), [WebBluetoothDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothDemo.js), [WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js)

Branch | Build ID | Date | Result 
---|---|---|---
master | 5844f9f | Nov 28, 2016 | FAIL
master | 6037375 | Nov 28, 2016 | PASS

jprestwo, 2016-11-28:
https://github.com/01org/zephyr.js/pull/483 should fix this.

xiaolongx, 2016-11-30:
QA retest this issue after #483 is merged, two issues still occurs.
1. GPIO: openAsync does not work
[ButtonLEDsAsync.js](https://github.com/01org/zephyr.js/blob/master/samples/ButtonLEDsAsync.js), [test-gpio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gpio.js)

Serial Logs:
```
Async GPIO test with two LEDs and a button...
zjs_gpio_open: invalid pin
```
2. UART: no serial log output via /dev/ttyACM0
[UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)

jprestwo, 2016-12-01:
https://github.com/01org/zephyr.js/pull/494 fixes both GPIO and UART. The problem was with promises, which had broken after the callback optimization change.

grgustaf, 2016-12-06:
@xiaolongx, please retest.

xiaolongx, 2016-12-06:
This issue cannot reproduce with commit a162db1 on both Arduino_101 and FRDM_K64F, close

closed 2016-12-06 by xiaolongx