intel/zephyr.js #490: convert.sh doesn't like % in comments in PWM.js

grgustaf, 2016-11-30:
This is on line 82 of convert.sh. Probably it could be changed to:

```bash
echo -n $CHAR >> $OUTPUT
```

Also, it looks like below there's a `printf "\n"` that should probably be removed? It's not being echoed to the file, where it doesn't belong anyway AFAICT.

jprestwo, 2016-11-30:
@grgustaf, what is the results of running convert.sh normally? It seems convert and run the JS fine for me. The printf for the newline was put there for some reason, but I cant seem to remember why. I think it was for adding a new line after we write the percentage out, essentially a aesthetic thing.

grgustaf, 2016-12-06:
@jprestwo, just run "convert.sh samples/PWM.js out" and I think you will see the complaint in the first few lines of output, like I do.

Anyway, I just submitted a PR and you can review that.

grgustaf, 2016-12-06:
PR  #502 fixes this.

closed 2017-01-18 by grgustaf