intel/zephyr.js #500: [REG] Performance.now() result over known delay

xiaolongx, 2016-12-05:
The test result of [test-performance.js](https://github.com/01org/zephyr.js/blob/master/tests/test-performance.js) fail on both Arduino_101 and FRDM_K64F, the diff time is 1020 now in test-performance.js.

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino_101/FRDM_K64F | 5844f9f | Dec 5, 2016 | PASS
master | Arduino_101/FRDM_K64F | a162db1 | Dec 5, 2016 | FAIL

grgustaf, 2016-12-06:
Interesting, @pfalcon does this surprise you? Do you see the same thing?

pfalcon, 2016-12-06:
@grgustaf : Will try later today. But I guess we'll just need to bump the delta...

pfalcon, 2016-12-06:
Yes, I confirm that with the latest master (2279b157ad4fb0c12f5aa4323549844e7448c503) I get the failure.  I submitted https://github.com/01org/zephyr.js/pull/508 to address (well, work around) that. It would be nice to understand why it's regressed, but with the current extensive development of both Z and Z.js, I'd say that it's not the best application of effort. We can return to that later, when jitter/latency like that will actually affect real-world usecases.

grgustaf, 2016-12-07:
Makes sense, thanks.

xiaolongx, 2016-12-12:
This issue cannot reproduce on both Arduino 101 and FRDM_K64F with commit 31c3908.

closed 2016-12-12 by xiaolongx