intel/zephyr.js #501: Issues upgrading to the recent JerryScript upstream

pfalcon, 2016-12-05:
After https://github.com/01org/zephyr.js/pull/450 , the next logical step is to upgrade JerryScript which is quite behind the upstream (and so for example doesn't support all boards available in Zephyr). I did this in my fork, https://github.com/pfalcon/zephyr.js/commits/aeolus-2016.12 , but unfortunately, there're an issue: when building against such JrS upstream, there's an interactive question from Kconfig:

~~~
System clock's h/w timer frequency (SYS_CLOCK_HW_CYCLES_PER_SEC) [] (NEW) 
~~~

Default value for SYS_CLOCK_HW_CYCLES_PER_SEC is in 100% cases set by a particular SoC's config (e.g. arch/nios2/soc/nios2f-zephyr, arch/arc/soc/em11d, arch/x86/soc/atom, arch/arm/soc/atmel_sam3, arch/arm/soc/nordic_nrf5/nrf52, etc.), so it should never should be force-asked from a user. If it is, then apparently board/soc/arch isn't selected/passed properly on that Kconfig invocation. This needs further looking into.

All in all, I submit this issue to let everyone be in loop on it, and have a place to discuss/report progress on it.

Excerpt from build log leading to Kconfig issue:

~~~
  CC      src/zjs_timers.o
  CC      src/zjs_util.o
  CC      src/zjs_k64f_pins.o
  LD      src/built-in.o
  AR      libzephyr.a
Building libjerry-core.a
make -C /home/pfalcon/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/jerryscript -f targets/zephyr/Makefile.zephyr BOARD=frdm_k64f jerry
targets/zephyr/Makefile.zephyr:47: build/frdm_k64f/zephyr/Makefile.export: No such file or directory
make -f ./targets/zephyr/Makefile O="build/frdm_k64f/zephyr" V= USER_LIBS="jerry-core" USER_LIB_INCLUDE_DIR="-L /mnt/hdd/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/jerryscript/build/frdm_k64f/obj-frdm_k64f/lib" TARGET_ZEPHYR=./targets/zephyr initconfig
Using /home/pfalcon/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig as base
Merging /home/pfalcon/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/zephyr/kernel/configs/unified.config
Merging /mnt/hdd/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/prj.conf
#
# configuration written to .config
#
make -f ./targets/zephyr/Makefile O="build/frdm_k64f/zephyr" V= USER_LIBS="jerry-core" USER_LIB_INCLUDE_DIR="-L /mnt/hdd/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/jerryscript/build/frdm_k64f/obj-frdm_k64f/lib" TARGET_ZEPHYR=./targets/zephyr outputexports
scripts/kconfig/conf --silentoldconfig Kconfig
.config:81:warning: symbol value '' invalid for SYS_CLOCK_HW_CYCLES_PER_SEC
*
* Restart config...
*
*
* General Kernel Options
*
System tick frequency (in ticks/second) (SYS_CLOCK_TICKS_PER_SEC) [100] 100
System clock's h/w timer frequency (SYS_CLOCK_HW_CYCLES_PER_SEC) [] (NEW) 
~~~

pfalcon, 2016-12-05:
@daniel-thompson: I'm afraid I need to call you up on this. So, the issue happens on "make output exports". It's called like (wrapped for readability):

~~~
make -f ./targets/zephyr/Makefile O="build/frdm_k64f/zephyr" \
V= \
USER_LIBS="jerry-core" \
USER_LIB_INCLUDE_DIR="-L /mnt/hdd/projects-3rdparty/Embedded/Zephyr/linaro-aeolus/zephyr.js/deps/jerryscript/build/frdm_k64f/obj-frdm_k64f/lib" \
TARGET_ZEPHYR=./targets/zephyr outputexports
~~~

So, it asks to do outputexports operation against O="build/frdm_k64f/zephyr", but outputexports appear to ignore it and somehow ends up doing its stuff in Zephyr's top directory.

I keep looking into this, but maybe you'd know what's wrong (how to do it right) already.

daniel-thompson, 2016-12-05:
Hi Paul

There's nothing wrong with the invocation. Its identical to the working invocation from the "real" JerryScript build system.

However its *very* odd to be calling initconfig again *after* building libzephyr.a . The config must be fully initialized otherwise you can't build the library, nothing should need to call initconfig again here. So I suggest you go back a few lines and figure out what is going in there first!

Also can you gather the same logs with V=1?

Daniel.

pfalcon, 2016-12-05:
Ok, the problem can be approached from the other end: JerryScript itself builds fine, so the problem must be in how Zephyr.js calls it to build. And well, it uses @nashif's way to build 3rd-party libs, as examplified by https://github.com/nashif/zephyr_modules. With this approach, Zephyr calls 3rd-party build system from its (Zephyr's) context.

By there was a counter-argument against that approach (discussed in https://jira.zephyrproject.org/browse/ZEP-366 and patches linked from it): 3rd-party build systems may be arbitrary complex, and it may be cumbersome to try to integrate them "into Zephyr". On the other hand, Zephyr's build system while complex, from our point of view it's at least "finitely complex", so using "inversion of control" approach would be better, where 3rd-party build system calls Zephyr to: 1) get build config (CC, CFLAGS); 2) build Zephyr source; 3) perform final link, with 3rd-party app injecting its own libs into it. (Steps 2&3 can be a single stage.)

This approach is now fully implemented in Zephyr upstream (https://gerrit.zephyrproject.org/r/#/c/4915/) and even documented there (https://gerrit.zephyrproject.org/r/6951). That's also what projects like JerryScript and MicroPython use.

Back to the topic of this ticket, Zephyr, via Makefile.app, calls JrS build system. But that's not how it's expected to be called, it expects to call Zephyr itself. When it does so (it's recursive call back into Zephyr build system thus), something goes wrong.

So, the solution is to call JrS makefile directly, instead of "passing it as a callback" to Zephyr. This seems to work well, after additional testing I'll submit a patch.

And indeed, it calls it pretty interesting way: it uses undocumented (within bounds of Zephyr tree) magic called "Makefile.app" which was @nashif's try on supporting linking 3rd-party libs with Zephyr.

pfalcon, 2016-12-05:
@daniel-thompson : Thanks for the reply, seems like we posted comments at the same time, except I posted mine to the wrong ticket first ;-). So, the issue seems to be understood and likely resolved (needs more testing).

grgustaf, 2016-12-06:
Sounds great @pfalcon, look forward to seeing the patch.

pfalcon, 2016-12-06:
Patch posted: https://github.com/01org/zephyr.js/pull/509.

grgustaf, 2018-01-08:
Fixed long ago.

closed 2018-01-08 by grgustaf