intel/zephyr.js #503: Unable to allocate buffer after pressing "Enter" key several times in UART.js

xiaolongx, 2016-12-06:
Press "Enter" key several times in [UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js) with commit a162db1, it seems hang in /dev/ttyACM0.

Error logs:
```
Starting UART example...
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_promise.c:160 zjs_fulfill_promise(): native handle not found
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
zjs_buffer: unable to find string buffer
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_promise.c:160 zjs_fulfill_promise(): native handle not found
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_buffer.c:358 zjs_buffer_create(): zjs_buffer_create: unable to allocate buffer
```

jprestwo, 2016-12-22:
@grgustaf, not sure if you remember our brief discussion of this, but the reason this is happening is because we create a new buffer for each character that gets read in. This causes too many buffers to be created and it gets to its limit before garbage collection can clean them up. The only way I can see of "fixing" this, is to change the API to use regular strings instead of buffers. Then we can just keep a single internal buffer for all the read data. Other than that, I don't know what we can do. If you change the read range min to something higher it helps, but then the sample will not work because it relies on reading in one character at a time (to process new-line and carriage return).

xiaoyu3x, 2017-01-17:
This issue cannot reproduce on arduino 101 with  commit c707669

closed 2017-01-17 by xiaolongx