intel/zephyr.js #504: [REG][Sensor] Catch errors when running Accelerometer.js

xiaolongx, 2016-12-06:
Catch errors when running [Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js), this issue was fixed with #483, but reproduce again.

Error logs
```
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_callbacks.c:332 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
S/ww50/zephyr.js/sracceleration (m/s^2):  x=0.004788 y=0.004788 z[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_callbacks.c:332 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
phyr.js/sracceleration (m/s^2):  x=0.004788 y=0.004788 z=0[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_callbacks.c:332 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
[ERROR] /home/xiaolong/00_xiaolong/ZJS/ww50/zephyr.js/src/zjs_callbacks.c:332 zjs_signal_callback(): error putting into ring buffer, ret=4294967174
```

Branch | Build ID | Date | Result
---|---|---|---
master | 870d928 | Nov 15, 2016 | FAIL
master | 103ca87 | Nov 30, 2016 | PASS
master | a162db1 | Dec 6, 2016 | FAIL

jimmy-huang, 2016-12-07:
this bug appears to be a completely different issue than last one, this one seems to have been introduced by recent upgrade to the unified kernel in Zephyr, I just tried the acceleometer sample in zephyr master, and it appears it's completely broken as well, notice that it report 0.004788 on all axis, which is basically 0, 0, 0 in the zephyr sample.  I think we'll have to wait for zephyr to fix the bmi160 driver first, see https://jira.zephyrproject.org/browse/ZEP-1358

xiaolongx, 2016-12-27:
No error messages display on Arduino 101, but it report 0.004788 on all axis.
Serial logs:
```
Accelerometer test...
ipm_console0: 'Sensor core running ZJS ARC support image'
state: activating
state: activated
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
acceleration (m/s^2):  x=0.004788 y=0.004788 z=0.004788
```

brianjjones, 2017-01-05:
This is likely a Zephyr bug https://jira.zephyrproject.org/browse/ZEP-1358

jimmy-huang, 2017-02-02:
two related bugs ZEP-1358 and ZEP-1421 are fixed upstream, however, the signal errors is a limitation of the our ringbuffer where it just can't handle too much signal events at a relatively fast rate like more than 30Hz, so I added a check in #604 where it will only fire event if values have changed, so hopefully this is fixed.

qiaojingx, 2017-02-15:
This issue is not reproduced on Arduino 101 with commit 3f8eb85.
Serial log:
```
Accelerometer test...
ipm_console0: 'Sensor core running ZJS ARC support image'
state: activating
state: activated
acceleration (m/s^2):  x=0.148428 y=0.033516 z=78.33168
acceleration (m/s^2):  x=0.110124 y=0.09576 z=9.834552
acceleration (m/s^2):  x=0.105336 y=0.105336 z=9.901584
acceleration (m/s^2):  x=0.114912 y=0.086184 z=9.86328
acceleration (m/s^2):  x=0.100548 y=0.090972 z=9.882432
acceleration (m/s^2):  x=0.1197 y=0.090972 z=9.872856
```

closed 2017-02-15 by xiaolongx